lvgl/lv_arduino #2: ESP32 - lv_task_handler() crashes

Pablo2048, 2019-02-17:
Hi @kisvegabor ,
thank You for this perfect GUI library. Unfortunately I've got issue to get even the demo working with latest lvgl. It seems like call to lv_task_handler() crashes the CPU. I've turned on tracing and put some additional dumps into lvgl and it seems like the crash is in lv_refr.c on line `obj->design_func(obj, &obj_ext_mask, LV_DESIGN_DRAW_MAIN);` at which CPU ends with:
[image]
It seems like it is same problem like https://github.com/littlevgl/lvgl/issues/431 , unfortunately I'm not able to solve this out. Latest working version of lvgl on ESP32 with Arduino demo is https://github.com/littlevgl/lvgl/releases/tag/v5.1.1 . Can You please advice, what can I try to help solve this out? Thank You.

kisvegabor, 2019-02-19:
Thanks for the report. 

`obj->design_func` is still called? If not can you print the value of `obj`?

Pablo2048, 2019-02-21:
Ok, so I started over with fresh copy of lvgl master and just modified lv_refr.c with following
```
  char buf[256];
  sprintf(buf, "LVGL: design_func: obj=%p, obj->design_func=%p", obj, obj->design_func);
  LV_LOG_TRACE(buf);
```
right before `obj->design_func(obj, &obj_ext_mask, LV_DESIGN_DRAW_MAIN);` and I've got this:
[image]
Then I've replaced lvgl with dev-6.0 branch just to try and got the  same error.

kisvegabor, 2019-02-21:
Do you use any compiler optimization? If yes can you try without it?

Pablo2048, 2019-02-21:
I'm using PlatformIO so I have modified my platofrmio.ini to
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_speed = 460800
build_unflags = -Os
build_flags =
  -O0
  -DCORE_DEBUG_LEVEL=5
  -DDEBUG_ESP_PORT=Serial
monitor_speed=115200
```
Code size has raised from 485476 before to 594316 bytes - so I think that optimisations are turned off. Unfortunately the crash persists. Now I'm searching for a way to turn off LTO (if it is enabled)...

kisvegabor, 2019-02-21:
I see. 

Form the log it seems you have an `lv_label` and an `lv_obj` object. Can trace further in `lv_obj_design` and `lv_label_design`?

Pablo2048, 2019-02-21:
Ok. I've put some LV_LOG_TRACEs and so far end with:
```
Starting...
lib\lvgl\src\lv_objx\lv_label.c@68->label create started
lib\lvgl\src\lv_core\lv_obj.c@206->Object create started
lib\lvgl\src\lv_objx\lv_label.c@133->label created
lib\lvgl\src\lv_misc\lv_task.c@65->lv_task_handler started
lib\lvgl\src\lv_core\lv_refr.c@187->display refresh task started
lib\lvgl\src\lv_core\lv_obj.c@1805->lv_obj_design
lib\lvgl\src\lv_core\lv_obj.c@1809->lv_obj_design->LV_DESIGN_COVER_CHK
lib\lvgl\src\lv_core\lv_obj.c@1641->lv_obj_get_opa_scale
lib\lvgl\src\lv_core\lv_obj.c@1649->lv_obj_get_opa_scale - before return LV_OPA_COVER
lib\lvgl\src\lv_core\lv_refr.c@577->LVGL: design_func: obj=0x3ffc40e8, obj->design_fuc=0x400d30cc
lib\lvgl\src\lv_core\lv_obj.c@1805->lv_obj_design
lib\lvgl\src\lv_core\lv_obj.c@1836->lv_obj_design->LV_DESIGN_DRAW_MAIN
lib\lvgl\src\lv_core\lv_obj.c@1840->lv_obj_design after get_style
lib\lvgl\src\lv_core\lv_obj.c@1641->lv_obj_get_opa_scale
lib\lvgl\src\lv_core\lv_obj.c@1649->lv_obj_get_opa_scale - before return LV_OPA_COVER
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
```
so it seems like the cras is in`lv_draw_rect` ...

Pablo2048, 2019-02-21:
So i put debug outputs to lv_draw_rect.c and end with this:
```
lib\lvgl\src\lv_core\lv_refr.c@187->display refresh task started
lib\lvgl\src\lv_core\lv_obj.c@1805->lv_obj_design
lib\lvgl\src\lv_core\lv_obj.c@1809->lv_obj_design->LV_DESIGN_COVER_CHK
lib\lvgl\src\lv_core\lv_obj.c@1641->lv_obj_get_opa_scale
lib\lvgl\src\lv_core\lv_obj.c@1649->lv_obj_get_opa_scale - before return LV_OPA_COVER
lib\lvgl\src\lv_core\lv_refr.c@577->LVGL: design_func: obj=0x3ffc40e8, obj->design_fuc=0x400d310c
lib\lvgl\src\lv_core\lv_obj.c@1805->lv_obj_design
lib\lvgl\src\lv_core\lv_obj.c@1836->lv_obj_design->LV_DESIGN_DRAW_MAIN
lib\lvgl\src\lv_core\lv_obj.c@1840->lv_obj_design after get_style
lib\lvgl\src\lv_core\lv_obj.c@1641->lv_obj_get_opa_scale
lib\lvgl\src\lv_core\lv_obj.c@1649->lv_obj_get_opa_scale - before return LV_OPA_COVER
lib\lvgl\src\lv_draw\lv_draw_rect.c@68->lv_draw_rect
lib\lvgl\src\lv_draw\lv_draw_rect.c@72->lv_draw_rect - after get_height/width
lib\lvgl\src\lv_draw\lv_draw_rect.c@80->lv_draw_rect - after draw_shadow
lib\lvgl\src\lv_draw\lv_draw_rect.c@119->lv_draw_rect_main_mid
lib\lvgl\src\lv_draw\lv_draw_rect.c@133->lv_draw_rect_main_mid - after draw_cont_radius_corr
lib\lvgl\src\lv_draw\lv_draw_rect.c@156->lv_draw_rect_main_mid - before fill_fp1
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
```
Crash is in `fill_fp(&work_area, mask, mcolor, opa);`
My debug is as
```
        LV_LOG_TRACE("lv_draw_rect_main_mid - before fill_fp1");
        fill_fp(&work_area, mask, mcolor, opa);
        LV_LOG_TRACE("lv_draw_rect_main_mid - after fill_fp1");
```
and after each Serial.print() is 100ms delay to allow serial transfer to finish.

kisvegabor, 2019-02-21:
It 's a big step! `fill_fp` a function pointer to `lv_vfill` in `lv_draw_vbasic.c`. Can you still narrow the location of the issue?

Pablo2048, 2019-02-21:
Ok - little bit further:
```
lib\lvgl\src\lv_draw\lv_draw_rect.c@133->lv_draw_rect_main_mid - after draw_cont_radius_corr
lib\lvgl\src\lv_draw\lv_draw_rect.c@156->lv_draw_rect_main_mid - before fill_fp1
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@123->lv_vfill
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@132->lv_vfill - after vdb_get
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@144->lv_vfill - after area_intersect
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@232->lv_vfill - before sw_color_fill
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```
in `lv_draw_vbasic.c` I've put this:
```
    LV_LOG_TRACE("lv_vfill - before sw_color_fill");
    sw_color_fill(&vdb_p->area, vdb_p->buf, &vdb_rel_a, color, opa);
    LV_LOG_TRACE("lv_vfill - after sw_color_fill");
```
I have disabled USE_LV_GPU (too complicated code in lv_draw_vbasic instead for simple sw_color_fill :-) )

kisvegabor, 2019-02-21:
Please print the values of `vdb_p->buf` and `vdb_p->area.x1/y1/x2/y2`.

Pablo2048, 2019-02-21:
Here You are:
```
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@123->lv_vfill
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@132->lv_vfill - after vdb_get
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@144->lv_vfill - after area_intersect
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@232->lv_vfill - before sw_color_fill
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@235->vdb_p->buf 0x3ffc03c8, vdb_p->area X1:0, Y1:0, X2:319, Y2:23
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@597->sw_color_fill
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@606->sw_color_fill - after disp_get_active
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@610->sw_color_fill - has driver.vdb_wr
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

Pablo2048, 2019-02-21:
So another buch of info:
```
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@232->lv_vfill - before sw_color_fill
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@235->vdb_p->buf 0x3ffc03c8, vdb_p->area X1:0, Y1:0, X2:319, Y2:23
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@597->sw_color_fill
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@606->sw_color_fill - after disp_get_active
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@610->sw_color_fill - has driver.vdb_wr
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@614->sw_color_fill - before disp->driver.vdb_wr
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```
In `sw_color_fill` i have:
```
                LV_LOG_TRACE("sw_color_fill - before disp->driver.vdb_wr");
                disp->driver.vdb_wr((uint8_t *)mem, mem_width, col, row, color, opa);
                LV_LOG_TRACE("sw_color_fill - after disp->driver.vdb_wr");
```

kisvegabor, 2019-02-21:
I assume you haven't initialized `disp->driver.vdb_wr`  when you registered the display driver. It should be `NULL` by default and the library shouldn't use it.

What is the value of `disp->driver.vdb_wr` ?

Pablo2048, 2019-02-21:
Well, actually I have used lvgl_test.ino from example template which contains:
```
    /*Initialize the display*/
    lv_disp_drv_t disp_drv;
    disp_drv.disp_flush = my_disp_flush;
    lv_disp_drv_register(&disp_drv);
```
Value is:
```
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@597->sw_color_fill
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@606->sw_color_fill - after disp_get_active
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@612->sw_color_fill - has driver.vdb_wr @ 0x400d1098
lib\lvgl\src\lv_draw\lv_draw_vbasic.c@615->sw_color_fill - before disp->driver.vdb_wr
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```
When I add setting of vdb_wr like this:
```
    /*Initialize the display*/
    lv_disp_drv_t disp_drv;
    disp_drv.disp_flush = my_disp_flush;
    disp_drv.vdb_wr = NULL;
    lv_disp_drv_register(&disp_drv);
```
every seems to be working now... Thank You. Should I do PR and correct the Arduino template?

kisvegabor, 2019-02-21:
Ahhhh. `lv_disp_drv_init(&disp_drv);` is missing. 

It should look like this:
```c

    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);     
    disp_drv.disp_flush = my_disp_flush;
    lv_disp_drv_register(&disp_drv);
```

The Arduino issue was reported so many times and all because of it...
Thank you very much for tracing it! 

I updated the `ino` file: https://github.com/littlevgl/arduino/blob/master/lvgl_test/lvgl_test.ino

Pablo2048, 2019-02-21:
Thank You very much for the help...

closed 2019-02-21 by kisvegabor