lvgl/lv_arduino #4: ESP32 ili9341 display not working correctly (maybe disp_flush problem)

nikisalli, 2019-03-16:
With the example sketch loaded, 7/8 of the display is filled in a white/black stripy pattern while the other part seems to load the correct image piece by piece because it is always white (that is the correct background color in the example sketch) and with debugging messages on (so with a 100ms delay) i can see a part of the Hello World string appearing for just a frame.

The espi library settings are correct because i used it before and tried some sketches too to confirm that.

The lv_conf.h is just the standard one with debug enabled.

thank you in advance, (the library looks awesome!)

[image]

Pablo2048, 2019-03-16:
Hi,
that strippy pattern mean unwritten area. Please check SPI speed settings in TFT_eSPI library. Did you use unmodified example (just with correct TFT_eSPI settings) or do you have another peripheral on the same SPI bus?

nikisalli, 2019-03-16:
> Hi,
> that strippy pattern mean unwritten area. Please check SPI speed settings in TFT_eSPI library. Did you use unmodified example (just with correct TFT_eSPI settings) or do you have another peripheral on the same SPI bus?

Hi thank you for the answer,
i have no other devices on the spi bus, the display runs in 8bit bus mode, maybe that is the problem?

Pablo2048, 2019-03-16:
I've never tried to communicate with ILI9341 over parallel bus so i can't say. Can you please try this:
  * initialize the display
  * fill some memory area with solid color
  * call disp_flush and check, if the TFT behave correctly

kisvegabor, 2019-03-17:
Do you call `lv_flush_ready` when you finish `flushing?`

nikisalli, 2019-03-17:
It is the example code i just found in the examples directory:
https://github.com/littlevgl/lv_arduino/blob/master/examples/ESP32_TFT_eSPI/ESP32_TFT_eSPI.ino

Pablo2048, 2019-03-17:
Ok, so please try steps I've suggested and post results here...

nikisalli, 2019-03-17:
Yes, i have to try today to see what happens, thanks!

nikisalli, 2019-03-17:
ok i tried and nothing changes, only the top small part changes in color as expected.
i wanted to ask, is the stripy pattern something the display does or is the lvgl library doing it?

nikisalli, 2019-03-17:
anyways the problem is how disp_flush writes colors, writecolor() doesn't work correctly because the library works if i use drawPixel instead but i guess it is a bit slower

Pablo2048, 2019-03-17:
The strippy pattern is result from display init. If writecolor() doesn't work right then the problem is in TFT_eSPI library, or (maybe) in the initialisation part of the TFT_eSPI. Display has to be set to the 16 bit RGB565 color IIRC.

nikisalli, 2019-03-17:
ok thanks, i think i'll keep using drawpixel until i find something faster that actually works

closed 2019-03-17 by nikisalli

sonastyle, 2020-01-09:
I'm  getting the same issue, how did you solve that?