lvgl/lv_arduino #6: XPT2046 support

jmcastillejo, 2019-06-09:
I'm testing the library on an ili9488 screen with the xpt2046 touch controller.

The screen works fine, but the touch panel does not work.

Any ideas? Regards

Pablo2048, 2019-06-09:
You have to write your own driver for touch panel - you can get some inspiration how to do it from here https://github.com/littlevgl/lv_drivers/blob/master/indev/XPT2046.c .

jmcastillejo, 2019-06-09:
The inspiration came to me suddenly!!

Replace the bool read_encoder function with this:

bool my_input_read(lv_indev_data_t *data)
{

  // Use TFT_eSPI for touch events
  uint8_t     bPressed = 0;
  uint16_t    nX=0;
  uint16_t    nY=0;
  static uint16_t prev_x, prev_y;
  
  bPressed = tft.getTouch(&nX,&nY);

    if (bPressed > 0) {
    data->point.x = nX;
    data->point.y = nY;
    data->state = LV_INDEV_STATE_PR;
    prev_x = data->point.x;
    prev_y = data->point.y;
    } else {
    data->point.x = prev_x;
    data->point.y = prev_y;
    data->state = LV_INDEV_STATE_REL;      
    }

  return false;        /*No buffering so no more data read*/
}

Replace the touchpad startup routine with this:

  /*Initialize the touch pad*/

// calibration touch pad
  uint16_t m_anCalData[5] = { 253, 3651, 235, 3597, 1 };
  tft.setTouch(m_anCalData);
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read = my_input_read;
  lv_indev_drv_register(&indev_drv);

closed 2019-07-07 by stale[bot]