lvgl/lv_arduino #7: `src` directory needs to made into a submodule

embeddedt, 2019-06-17:
Including LittlevGL as part of a different repository's history is a bad thing.

The `src` directory should be converted to a submodule. It should be as simple as removing the directory and adding it again with `git submodule add https://github.com/littlevgl/lvgl src`.

embeddedt, 2019-06-17:
@Pablo2048 Does the Arduino IDE know how to handle submodules?

Pablo2048, 2019-06-17:
Well, unfortunately not - Arduino IDE won't work with any VCS system. Also working with submodules in Arduino library is IMHO no go - library is tight with LVGL version (asctually 5.X IIRC).

embeddedt, 2019-06-17:
@Pablo2048 Okay. Forgive my ignorance, but what are these lines for?

https://github.com/littlevgl/lv_arduino/blob/4ca926a526cb4399c7e809df980375b9f6bad975/library.json#L10-L14

Pablo2048, 2019-06-17:
:-) theyr'e for using with Library Manager. Library manager is just very simple method to get the library from (for example github). So I was wrong with my "won't work..." explanation. But I was unable to get this to work with submodules - only with releases, and/or masters.

Pablo2048, 2019-06-17:
Moving Arduino files into LittleVGL root directory maybe get us to some troubles - source code of the library need to be in /src subdirectory, or in root directory. Examples should be in /examples folder.

embeddedt, 2019-06-17:
@Pablo2048 

When someone downloads this library, do they get the ESP32 drivers as well? I am helping a beginner over [here](https://forumtest.littlevgl.com/t/noob-support-for-ili9486/86/15) and I just want them to get LittlevGL itself.

> source code of the library need to be in /src subdirectory, or in root directory

The source code is all inside `src`, but in subdirectories (i.e. `src/lv_core`, `src/lv_objx`) - is that acceptable?

Pablo2048, 2019-06-17:
Arduino library folder structure is specified here https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification . I'm using rev 2.1 format for > 1.5.x version (without backward compatibility with 1.0.x) so top level source files and includes needs to be in src folder and not in subdirectories. Nested files from 'main" library include can be put into subdirectories. Example, given in examples is based on the TFT_eSPI  library used as an 'low level' driver, so TFT_eSPI library has to be installed and properly configured to match the LCD hardware prior compiling the example - as described in README.md

embeddedt, 2019-06-17:
@Pablo2048 Are custom include paths supported inside Arduino?

Pablo2048, 2019-06-17:
Partially yes - you have to use full absolute path or edit platform configuration file (which is both no-go IMHO) as described here https://forum.arduino.cc/index.php?topic=518223.0 . Arduino IDE is very simple tool - I suggest to use Visual Studio Code and Platformio. Install both, set the framework to Arduino and in platformio.ini file inside project directory there can be include paths configured.
Anyway - TFT_eSPI library seems to have support for 9486 - when configured properly there is no reason to don't get it to work (https://github.com/Bodmer/TFT_eSPI).

embeddedt, 2019-06-17:
@Pablo2048 I was trying to keep things simple because they are new to this type of development.

Do you think you could help them on the forum?

Pablo2048, 2019-06-17:
Yes - definitely I can help, but I'm in a rush until end of this week :-(

Pablo2048, 2019-07-08:
So I've searched for way to use submodule. Unfortunately I've found this:

> https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#are-git-submodules-supported

So no luck here, and I'm closing this...

closed 2019-07-08 by Pablo2048