lvgl/lv_arduino #9: V6 how to use it with arduino ?

luc-github, 2019-07-03:
Littlevgl is amazing - I have started recently with v5.3 
Now littlevgl V6 is out - how to use it with arduino ?

Just replace scr directory with latest version ? Any recommendation ?

embeddedt, 2019-07-03:
@luc-github I would recommend waiting - @Pablo2048 is working on updating the project already.

luc-github, 2019-07-03:
@embeddedt @luc-github  thank you - let me know if I can help in anything ^_^

Pablo2048, 2019-07-04:
Ok, I'm almost done. The only thing that remains are font files with ```#include "lvgl/lvgl.h"``` - I'm unable to force Arduino to compile this so far (Arduino Builder is unable to locate the lvgl subdirectory). When I change ```#include "lvgl/lvgl.h"``` to ```#include "lvgl.h"``` it compiles ok...

kisvegabor, 2019-07-04:
@Pablo2048 I've updated the built-in fonts to use relative path to `lvgl.h`

Pablo2048, 2019-07-04:
Great - thanks. So I use current master instead of Release. Is this ok? If all my testing went ok, I will do release new version of the library until end of this week.

kisvegabor, 2019-07-04:
> So I use current master instead of Release Is this ok?

Yes. We continuously add small fixes to master.

Pablo2048, 2019-07-07:
Library updated with the new core. I'm not doing an release for now, because it's not based on actual LVGL release - I'm using master...

luc-github, 2019-07-07:
Thank you so much ^_^ 
So lv_arduino release will be when 6.1 is released, I am right ?
Is there a way to automate the creation process when a lvgl release will happen,  without the example part of course, because this one need human action if API change

Pablo2048, 2019-07-07:
Yes. We have two options:
1. wait for next lvgl 6.x release (I'm preferring this solution)
2. release v2.0.0 with Master branch from lvgl and then v2.0.1 after new lvgl 6.x release

I'm searching for a way to do some automation, but my github-fu seems to be too weak :-( so if anybody can give me a hand...

luc-github, 2019-07-07:
about automation I would be happy to help - I think the idea would be:
Once you have a script that automate the task :
* you can use travis and the 
`- if [ -n "$TRAVIS_TAG" ]; then something when tag set; fi` feature to launch this script (I use it on one project to generate binary when doing a release
) 
* and then use this feature also to push your commit to lv_arduino repository and/or  any other repository using thisfor example :
https://gist.github.com/Maumagnaguagno/84a9807ed71d233e5d3f

embeddedt, 2019-07-07:
The master branch is always kept stable -  I would suggest releasing now.

kisvegabor, 2019-07-08:
@Pablo2048 Maybe I missed the explanation but why can we have an Arduino release for v6.0?

Pablo2048, 2019-07-08:
@kisvegabor Arduino Library manager work only with Releases so without doing an lv_arduino release, users have to install the library manually.
@luc-github Thank You - I'm new to travis so I need some time to get familiar with it
@embeddedt Ok, if there is need, I'll do an release for version 2.0.0

Pablo2048, 2019-07-08:
Release v 2.0.0 done...

luc-github, 2019-07-08:
thank you ^_^

closed 2019-07-08 by luc-github

kisvegabor, 2019-07-08:
Awesome! :slightly_smiling_face: