lvgl/lv_arduino #11: Cant find lv_event_t

shr1Khr, 2019-07-07:
So im trying to trigger an event from the button push, and compiler says - 
 error: 'lv_event_t' has not been declared
I tried searching for this structure in the core files, not there. Help!

Here is the triggering point - 
`lv_obj_set_event_cb(MenuBtn, list_btn_event_handler);            // Assign an event callback`

And here is the recieving function which is throwing error-
`
static void list_btn_event_handler(lv_obj_t * btn, lv_event_t event){
`
`switch(event) {
    case LV_EVENT_SHORT_CLICKED:
      printf("Pressed\n");
      break;
}
`

Pablo2048, 2019-07-07:
Hi,
```lv_event_t``` is introduced in lvgl 6.0. lv_arduino is based on older 5.3 release which doesn't have this. I'm working on library update...

Pablo2048, 2019-07-07:
New version of the library is on github (not released for now). You can give it a try...

shr1Khr, 2019-07-07:
Thanks for the pointer, I'll give it a shot and will update here, cheers.

shr1Khr, 2019-07-07:
It works, I see a lot of things changed in v6, but all good, Thanks!

closed 2019-07-07 by shr1Khr