lvgl/lv_arduino #13: ESP32 with lv_aruino  can't set label style.

TridentTD, 2019-07-10:
I have just tried the example "ESP32_TFT_eSPI"  from the current master lv_arduino, 
it can run fine.

However when I want to change label style to red color
by the following code. It occure error at run-time.

```C
#include <lvgl.h>
#include <Ticker.h>
#include <TFT_eSPI.h>

#define LVGL_TICK_PERIOD 20

Ticker tick; /* timer for interrupt handler */
TFT_eSPI tft = TFT_eSPI(); /* TFT instance */
static lv_disp_buf_t disp_buf;
static lv_color_t buf[LV_HOR_RES_MAX * 10];

#if USE_LV_LOG != 0
/* Serial debugging */
void my_print(lv_log_level_t level, const char * file, uint32_t line, const char * dsc)
{

  Serial.printf("%s@%d->%s\r\n", file, line, dsc);
  delay(100);
}
#endif

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint16_t c;

  tft.startWrite(); /* Start new TFT transaction */
  tft.setAddrWindow(area->x1, area->y1, (area->x2 - area->x1 + 1), (area->y2 - area->y1 + 1)); /* set the working window */
  for (int y = area->y1; y <= area->y2; y++) {
    for (int x = area->x1; x <= area->x2; x++) {
      c = color_p->full;
      tft.writeColor(c, 1);
      color_p++;
    }
  }
  tft.endWrite(); /* terminate TFT transaction */
  lv_disp_flush_ready(disp); /* tell lvgl that flushing is done */
}

/* Interrupt driven periodic handler */
static void lv_tick_handler(void)
{

  lv_tick_inc(LVGL_TICK_PERIOD);
}

/* Reading input device (simulated encoder here) */
bool read_encoder(lv_indev_drv_t * indev, lv_indev_data_t * data)
{
  static int32_t last_diff = 0;
  int32_t diff = 0; /* Dummy - no movement */
  int btn_state = LV_INDEV_STATE_REL; /* Dummy - no press */

  data->enc_diff = diff - last_diff;;
  data->state = btn_state;

  last_diff = diff;

  return false;
}

void setup() {

  Serial.begin(115200); /* prepare for possible serial debug */

  lv_init();

#if USE_LV_LOG != 0
  lv_log_register_print(my_print); /* register print function for debugging */
#endif

  tft.begin(); /* TFT init */
  tft.setRotation(1); /* Landscape orientation */

  lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);

  /*Initialize the display*/
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = 320;
  disp_drv.ver_res = 240;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.buffer = &disp_buf;
  lv_disp_drv_register(&disp_drv);

  /*Initialize the touch pad*/
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_ENCODER;
  indev_drv.read_cb = read_encoder;
  lv_indev_drv_register(&indev_drv);

  /*Initialize the graphics library's tick*/
  tick.attach_ms(LVGL_TICK_PERIOD, lv_tick_handler);

  /* Create simple label */
  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino! (V6.0)");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);

  //---------------- My test for changing label's style--------------------------------
  lv_style_t label_style; 
  lv_style_copy(&label_style, &lv_style_plain);  
  lv_style_copy.text.color = LV_COLOR_RED;
  lv_obj_set_style( label,  &label_style);
  //-------------------------------------------------------------------------------------
}

void loop() {

  lv_task_handler(); /* let the GUI do its work */
  delay(5);
}
```

How to fix this issue?

Thank you very much.

Pablo2048, 2019-07-10:
You have to decode the stack trace - without decoding the stack trace is useless. Maybe also @kisvegabor can give us some advice after looking to the source code. And also maybe this is already fixed bug in last few days - lv_arduino is (unfortunately) not based on release, but master which get updates almost every day...

TridentTD, 2019-07-10:
OK, I have already fixed this issue.

I moved   "lv_style_t label_style;" 
to global variable declare. It can be fixed.

Thank you very much.

Pablo2048, 2019-07-10:
Ok, I'm closing this then...

closed 2019-07-10 by Pablo2048