lvgl/lv_arduino #14: lv_examples:  Arduino directory structure

Uup115, 2019-07-12:
Hello,

A previous version of "lv_test_theme_1" (January 2019) compiles successfully with lv_arduino (v1.0),  However, several hours of manipulation, to the **#include paths and directory structure**, were necessary.

I recently updated to lv_arduino (v2.0).  Despite diligent efforts, the current version "lv_test_theme_1" fails to compile.  

Please, can you let us know how to configure the lv_examples for Arduino?  Thanks.

Best Regards

Pablo2048, 2019-07-14:
Hi,
unfortunately it seems like lv_examples are not prepared to be used in Arduino. Relative paths to lvgl are no-go because of Arduino builder limits.

Uup115, 2019-07-15:
Hello,

Good news!  I successfully compiled the "lv_test_theme_1" example with  lv_arduino (v2.0).  

Basically, all relevant source files, **lv_test_theme_1.h** , **lv_test_theme_1.c**, and **lv_ex_conf.h** should reside in the same directory as the sketch file.  The #include paths, found in "lv_test_theme_1.h", should be changed to the following:

**#include <lvgl.h>
#include "lv_ex_conf.h"**

Arduino compiler doesn't seem to handle nested paths very well.

Best Regards

Pablo2048, 2019-07-19:
I'm closing this, because there is nothing what we can do in this library.

closed 2019-07-19 by Pablo2048