lvgl/lv_arduino #15: ddlist returns 0 always in arduino, esp32

shr1Khr, 2019-07-14:
I'll perhaps upload the sketch if you want, but i'm using standard code of [https://docs.littlevgl.com/en/html/object-types/ddlist.html](url)
and it returns 0 in the ddlist always, meaning ` lv_ddlist_get_selected_str(obj, buf, sizeof(buf));` returns the first option always.

Please check?

embeddedt, 2019-07-14:
Are you using a mouse-like device (mouse/touchscreen) or a keyboard-like device?

shr1Khr, 2019-07-14:
It's on tft screen with touch

embeddedt, 2019-07-14:
You are using this example? https://docs.littlevgl.com/en/html/object-types/ddlist.html#c

shr1Khr, 2019-07-14:
Yes, exactly like the example

embeddedt, 2019-07-14:
I don't need the whole sketch, but it would be easier if I could see the relevant code.

shr1Khr, 2019-07-14:
Here is the recieve function - 

`static void WifiScan_ddlist_cb(lv_obj_t * obj, lv_event_t event){
    if((event == LV_EVENT_VALUE_CHANGED) || event > 10) {
        char buf[32];
        lv_ddlist_get_selected_str(obj, buf, sizeof(buf));
        Serial.println(lv_ddlist_get_selected(ddlist_w));
        printf("Option: %s\n", buf);
    }
}`

and here is the trigger function - 

`ddlist_w = lv_ddlist_create(parent, NULL);   
    lv_ddlist_set_options(ddlist_w, "op1\nop2\nop3\0");
    lv_ddlist_set_fix_width(ddlist_w, 135);
    lv_ddlist_set_draw_arrow(ddlist_w, true);
    lv_obj_align(ddlist_w, NULL, LV_ALIGN_IN_TOP_MID, 40, 30);
    lv_obj_set_event_cb(ddlist_w, WifiScan_ddlist_cb);`

If you wanna see the video of whats happening, here it is - 
[https://www.youtube.com/watch?v=77nqofAcgaI](url)

My big fingers are not the problem (lol), I'm able to use the full keyboard on this display and touch.
Here is the serial output - 
Options found: 3 for X⸮⸮?H 
Options found: 3 for X⸮⸮?H 
0
Send 0 Option: op1
0
Send 0 Option: op1
0
Send 0 Option: op1
0
Option: op1
0
Option: op1
0
Send 0 Option: op1
0
Send 0 Option: op1
0
Send 0 Option: op1
0
Send 0 Option: op1
0
Send 0 Option: op1

I added the printf function in the src code to debug why its not working, and it is spitting out this - Options found: 3 for X⸮⸮?H (dont mind the "X⸮⸮?H" part, I passed object in serial print).
"Send 0 Option: op1"  is the send event part of src code, which shows the root cause, I think. That printf function is between line 789 and 790, here - [https://github.com/littlevgl/lv_arduino/blob/ac94d7c26258bcafc2a72d7976059b5b4f6073a0/src/src/lv_objx/lv_ddlist.c#L790](url)

Hope it helps

embeddedt, 2019-07-14:
I can see from the video that it never actually selects "op3".

It must be an issue with your touchscreen driver because the same code works fine for me on my board. Two things:

* Are you returning `false` from your `read_cb` function?
* When the touchscreen is released, do you pass the last pressed coordinates to `data->point`?

shr1Khr, 2019-07-14:
Yep to both the quesitons, here is my read_cb function - 

`bool read_encoder(lv_indev_drv_t * indev, lv_indev_data_t * data) {
  getTouchPoints();
  if (digitalRead(TINT) == 0) {
    punchNow();
    data->point.x = pt_touched[0];
    data->point.y = pt_touched[1];
    data->state = LV_INDEV_STATE_PR;
  }
  else {
    data->point.x = Prev_pt_touched[0];
    data->point.y = Prev_pt_touched[1];
    data->state = LV_INDEV_STATE_REL;
  }
  return false;
}`

getTouchPoints(); loads the variables pt_touched & Prev_pt_touched;
punchNow(); is prevent the esp32 from sleeping.

The point is, I'm able to swipe and do all sorts of touch things, but this is not working for some reason.
Can it be passing the "\n" as string for C++ & C ? Initially, I was struggling to pass the correct format of string, and I'm still not sure about it, but thought the standard code will work fine.

shr1Khr, 2019-07-14:
Your observation - I can see from the video that it never actually selects "op3". is actually the problem, so I was printing everything in event function, just to see of anytime at all it returns any other id. 
If the whole options string is interpreted as one object? and its displaying in different line because on \n ?

shr1Khr, 2019-07-14:
so, here is another video showing the response from the read_encoder function. (I copied it from the arduino example, hence named encoder, but its touch actually)
`https://www.youtube.com/watch?v=tKU5NHwbBVo`

embeddedt, 2019-07-14:
@shr1Khr It's very, very likely to be an issue with your platform, as the exact same code works properly on a different system.

Maybe you should try it using a bigger font (if you have the flash) so that accuracy becomes less of an issue?

shr1Khr, 2019-07-14:
sorry if its a dumb question, how do I set font for the list options?

embeddedt, 2019-07-14:
@shr1Khr Either change the style, or just change the global default font.

kisvegabor, 2019-07-15:
> It's very, very likely to be an issue with your platform, as the exact same code works properly on a different system.

I agree. You should also print the pressed coordinates in your touchpad read function to see if you really presse below the first option.

shr1Khr, 2019-07-15:
Sorry I was busy through out the day, I'll check the font thing tomorrow morning, will report the results here.
@kisvegabor I don't quite understand what you mean?

kisvegabor, 2019-07-15:
I mean, in the touchpad driver you should print which x/y coordinates you pressed and check if really there is a difference between
1. the press when you click the ddlist
2. the press when you select the option

The y coordinate should be greater when you select the option.

shr1Khr, 2019-07-15:
I have shown that in video#2 that I posted above, please check if that's not what you meant. The serial monitor in the background is printing x y coordinates of touch as I'm touching the screen.

kisvegabor, 2019-07-15:
Sorry, I didn't notice that. It seems correct.

An other thing which might help to see if the touchpad misbehaves is to create a dummy cursor:
```c
   ...
   lv_indev_t * mouse_indev = lv_indev_drv_register(&indev_drv);

   lv_obj_t * cursor_obj =  lv_label_create(lv_disp_get_scr_act(NULL), NULL); 
   lv_label_set_text(cursor_obj, "|");      
   lv_indev_set_cursor(mouse_indev, cursor_obj);

```

shr1Khr, 2019-07-16:
So the larger font didn't work, lemme try the cursor thing.

shr1Khr, 2019-07-16:
So I did check the cursor method, and I can see the touch is touching the different options, if you like to see, ill upload a video. However, the selected option is always the first one. 
Thoughts?

Uup115, 2019-07-17:
Hello shr1Khr,

I'm using the Arduino DUE + ILI9341 (320x240) Touch display + lv_Arduinio (v2.0).  The "ddlist" is working good for me.  No issues so far.

With lv_Arduino (v1.0), I experienced touch issues with ddlist.  It was very difficult to get the drop list to respond.  Many touches were required to activate the drop.  Similar to what you have experienced, the first option was always active, even when I selected another option.

Best Regards

embeddedt, 2019-07-17:
It is possible there is a bug in v5.3 (which v1 of the Arduino project uses). I only tested it on the  newer version.

shr1Khr, 2019-07-17:
Okay, umm I'll update everything, I thought I did, it's alright. 
The project I'm building requires users to select the units of the input values, I have to use the ddlist, because I can't ask user to enter the units manually as it would create coding nightmare. The scroll list requires more space on screen and maynot be feasible. I'll try and update everything, hopefully it will work, else, I'll look for alternatives. I need to get this thing sorted as I need to give it to users. Sorry, I'm not complaining, but sharing :)

shr1Khr, 2019-07-17:
So I just downloaded the arduino version of littlevgl, no luck. My arduino is 1.8.5. Im not sure why its not working. here is the new video. The cursor is also active, just hard to see in video.
[Link](https://www.youtube.com/watch?v=8KN1I5eF_RA)

Uup115, 2019-07-17:
Hello shr1Khr,

I'm using Arduino IDE v1.8.9.  I don't think this matters though.

Best Regards

embeddedt, 2019-07-17:
@shr1Khr Please confirm that `src/src/lv_version.h` contains:

```c
#define LVGL_VERSION_MAJOR   6
#define LVGL_VERSION_MINOR   0
#define LVGL_VERSION_PATCH   0
```

shr1Khr, 2019-07-17:
Yep, it is.

kisvegabor, 2019-07-18:
Could you share your `touchpad_read` function? Maybe we will spot the issue.

You can add some debug printfs into `release_handler` (lv_ddlist.c) to see what's going on.

shr1Khr, 2019-07-18:
Sure, it is bit messy, since I never re-visited it after getting it to work initially.

Here is the read_encoder function - 

```
bool read_encoder(lv_indev_drv_t * indev, lv_indev_data_t * data) {
getTouchPoints();
  if (digitalRead(TINT) == 0) {
    data->point.x = pt_touched[0];
    data->point.y = pt_touched[1];
    data->state = LV_INDEV_STATE_PR;
  }
  else {
    data->point.x = Prev_pt_touched[0];
    data->point.y = Prev_pt_touched[1];
    data->state = LV_INDEV_STATE_REL;
  }
  return false;
}
```

So I call initially getTouchPoints() in the read_encoder() to load pt_touched, it contains current touch points.

```
void getTouchPoints(){
 String tp = touch.getTouch();
  int len = tp.length();
  String yy = tp.substring(len-5);
  String xx = tp.substring(0, (len-5));
  if(Rotation == 2){
    pt_touched[0] = xx.toInt();
    pt_touched[1] = yy.toInt();
  }
  else if(Rotation == 0){
    pt_touched[0] = 240 - xx.toInt();
    pt_touched[1] = 240 - yy.toInt();
    }
}
```

Previous touch is fetched via 

```
void  clearTouchVar(){
  Prev_pt_touched[0] = pt_touched[0];
  Prev_pt_touched[1] = pt_touched[1];
  pt_touched[0] = 0;
  pt_touched[1] = 0;  
}
```

**Aaaaand guess what, this above function - clearTouchVar() - is never called!
So apparently Littlevgl  doesn't have a method to know last touch.
Suggestions where should I call this function?
Hope this is the root cause!**
I migrated from Adafruits graphics, where the above function was called in loop(), to clear the touch variables after the triggered actions were completed, but I commented everything since littlevgl.

Below is the touch driver, it also has the capability to read multiple touch, I never used it though. Hope if you guys can make use of below function in better way to be used with littlevgl.

```
void TouchDriver::touchDemo(){
  uint8_t press1, release1, contact1, touchID1, w1, m1, press2, release2, contact2, touchID2, w2, m2;
  uint16_t x1, y1, x2, y2;

    //Check to see if the interrupt is toggled. Since it's in polling mode, it will remain LOW 
    //whenever there is an active touch. If it is in trigger mode, it will pulse
    if (digitalRead(_tint) == 0){
      Wire.beginTransmission(FT62XX_ADDR);
      //Start reading at register 0x02
      Wire.write(0x02);
      //Send the data over
      Wire.endTransmission();

      Wire.requestFrom(FT62XX_ADDR, 16);
      //Wire.requestFrom(0x02, 16);

      //Register 2 contains the number of points touched
      uint8_t pointsTouched;
      pointsTouched = Wire.read() & 0x0F;

      if (0 < pointsTouched){
        //read the data
        ((uint8_t *)(&x1))[1] = Wire.read();
        ((uint8_t *)(&x1))[0] = Wire.read();
        ((uint8_t *)(&y1))[1] = Wire.read();
        ((uint8_t *)(&y1))[0] = Wire.read();
        w1 = Wire.read();
        m1 = Wire.read();

        //crunch the data
        press1 = ((x1 & 0xC000) == 0x0000) ? 1 : 0;
        release1 = ((x1 & 0xC000) == 0x4000) ? 1 : 0;
        contact1 = ((x1 & 0xC000) == 0x8000) ? 1 : 0;
        touchID1 = y1 & 0xE0000;
        x1 &= 0x0FFF;
        y1 &= 0x0FFF;
        String a = String(x1);
        String b = String(y1);
        temp_tp = a+"    "+b;
        
        //check for two points touched
        if (1 < pointsTouched) {
          ((uint8_t *)(&x2))[1] = Wire.read();
          ((uint8_t *)(&x2))[0] = Wire.read();
          ((uint8_t *)(&y2))[1] = Wire.read();
          ((uint8_t *)(&y2))[0] = Wire.read();
          w2 = Wire.read();
          m2 = Wire.read();
          press2 = ((x2 & 0xC000) == 0x0000) ? 1 : 0;
          release2 = ((x2 & 0xC000) == 0x4000) ? 1 : 0;
          contact2 = ((x2 & 0xC000) == 0x8000) ? 1 : 0;
          touchID2 = y2 & 0xE0000;
          x2 &= 0x0FFF;
          y2 &= 0x0FFF;
          Serial.println();   
          SerPrintFF(F(" (%5u,%5u) p=%3u r=%3u c=%3u ID=%3u \n"),
                          x2, y2, press2, release2, contact2, touchID2);
        }
    }
  }
}
```

PS - I finally googled how to add code in comments section of GitHub, like above

kisvegabor, 2019-07-18:
Maybe @Uup115 can help with `clearTouchVar`. Hopefully it will solve the issue

>  I finally googled how to add code in comments section of GitHub, like above

You even do it like
` ```c `
`... code ...`
` ``` `
to add syntax highlight.

Uup115, 2019-07-18:
Hello,

Why do you need to clear touch point data "clearTouchVar()" ?  Seems like a lot of code just to get the touch points.

I'm using a 3.2" ILI9341 serial display with touch:
[https://www.buydisplay.com/default/arduino-3-2-inch-tft-lcd-touch-shield-ili9341-library-for-mega-due-uno](url)

...and here is the callback code I use for touch ( [http://www.rinkydinkelectronics.com/library.php?id=92](url) ) :

`bool my_tp_read(lv_indev_drv_t * indev, lv_indev_data_t * data)
{  
   static lv_coord_t last_x = 0;
   static lv_coord_t last_y = 0;
   
   /*Save the state and save the pressed coordinate*/
    data->state = myTouch.dataAvailable() ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL; 
    if(data->state == LV_INDEV_STATE_PR){
      myTouch.read();
      last_x = (lv_coord_t)myTouch.getX();
      last_y = (lv_coord_t)myTouch.getY();
    }

    /*Set the coordinates (if released use the last pressed coordinates)*/
    data->point.x = last_x;
    data->point.y = last_y;

    return false; /*Return `false` because we are not buffering and no more data to read*/
}`

This above works nicely.

Best Regards,
Aaron

shr1Khr, 2019-07-19:
@Uup115 

So for the last pressed co-ordinates, you are passing 0 for x and y, true?

embeddedt, 2019-07-19:
@shr1Khr No, you are supposed to save the coordinates you get when the screen is pressed and pass those as the last pressed coordinates.

shr1Khr, 2019-07-19:
oh okay, so here is what I did -
Since I'm reading the touch points when touch interrupt == 0, old touch is stored in tp[] variable and used for LV_INDEV_STATE_REL.

Also, I tested and the ddlist is finally working!

Thanks guys for your help. Really appreciate it!

```
static uint16_t tp[2];

bool read_encoder(lv_indev_drv_t * indev, lv_indev_data_t * data) {
  if (digitalRead(TINT) == 0) {
    touch.gtp(tp);
    punchNow();
    data->point.x = (240 - tp[0]);                         // pt_touched[0];
    data->point.y = (240 - tp[1]);                         // pt_touched[1];
    data->state = LV_INDEV_STATE_PR;
  }
  else {
    data->point.x = (240 - tp[0]);                           // Prev_pt_touched[0];
    data->point.y = (240 - tp[1]);                           // Prev_pt_touched[1];
    data->state = LV_INDEV_STATE_REL;
  }
  return false;
}
```
And here is the function to get tp, I cleaned it - 

```
void TouchDriver::gtp(uint16_t * tp ){
  uint16_t x1, y1;
    if (digitalRead(_tint) == 0){
      Wire.beginTransmission(FT62XX_ADDR);
      Wire.write(FT_REG_NUM_FINGER);                // Start reading at register 0x02;
      
      Wire.endTransmission();                       // Send the data over

      Wire.requestFrom(FT62XX_ADDR, 16);            // Read 16 bytes from address 0x38
      Wire.requestFrom(FT_REG_NUM_FINGER, 16);
      
      uint8_t pointsTouched;                        // Register 2 contains the number of points touched
      pointsTouched = Wire.read() & 0x0F;

      if (0 < pointsTouched){
        ((uint8_t *)(&x1))[1] = Wire.read();
        ((uint8_t *)(&x1))[0] = Wire.read();
        ((uint8_t *)(&y1))[1] = Wire.read();
        ((uint8_t *)(&y1))[0] = Wire.read();
        x1 &= 0x0FFF;
        y1 &= 0x0FFF;
        tp[0] = x1;
        tp[1] = y1;  
    }
  }
}
```

closed 2019-07-19 by shr1Khr