lvgl/lv_arduino #18: lv_prelaod animation

Uup115, 2019-07-22:
Hello

The example code doesn't spin.  Please, how to enable the spin animation?  

https://docs.littlevgl.com/en/html/object-types/preload.html

Thanks

Pablo2048, 2019-07-23:
I was using this, but with 5.3 version
```C
// Boot up screen
  lv_obj_t *label = lv_label_create(mainCircle, NULL);
  lv_label_set_text(label, "Connecting to WiFi...");
  lv_obj_align(label, mainCircle, LV_ALIGN_CENTER, 0, -20);

  lv_style_copy(&preloaderstyle, &lv_style_plain);
  preloaderstyle.line.width = 6;                         /*?? px thick arc*/
  preloaderstyle.line.color = LV_COLOR_HEX3(0x258);       /*Blueish arc color*/

  preloaderstyle.body.border.color = LV_COLOR_HEX3(0xBBB); /*Gray background color*/
  preloaderstyle.body.border.width = 6;
  preloaderstyle.body.padding.hor = 0;

  /*Create a Preloader object*/
  lv_obj_t *preload = lv_preload_create(mainCircle, NULL);
  lv_obj_set_size(preload, 50, 50);
  lv_obj_align(preload, mainCircle, LV_ALIGN_CENTER, 0, 20);
  lv_preload_set_style(preload, LV_PRELOAD_STYLE_MAIN, &preloaderstyle);

  lvglMutex = xSemaphoreCreateMutex();
  xTaskCreate(gui_task, "lvgl", 10 * 1024, NULL, 5, NULL); // Start GUI task
```
I'll try to update the code to V6.0 and let you know...

kisvegabor, 2019-07-23:
I've tested it with the latest master branch (not in arduino) and it works for me.

@Uup115 Could you manually update lvgl from the latest master and try again?

Uup115, 2019-07-23:
@kisvegabor  Per your request, I updated from the latest master...  With the update, it makes only **one** complete CCW rotation, then stops.

embeddedt, 2019-07-23:
@Uup115 When it "stops", has the processor crashed, or does LittlevGL keep running?

Uup115, 2019-07-23:
@embeddedt  LittlevGL keeps running as normal.  No crashes.

embeddedt, 2019-07-23:
How is `lv_tick_inc` called on Arduino?

Uup115, 2019-07-23:
@embeddedt  Thanks for your last message, regarding "lv_tick_inc" .  I discovered an error in the way I configured the Timer interrupt.  Totally my fault.  Really, really sorry. :-(

I've already kicked myself several times.  I'll go ahead and kick myself several more times for you, kisvegabor, and Pablo2048.

Pablo2048, 2019-07-23:
:-) Ok, so it works now? Can you please try with pure lv_arduino? If it works only with actual sources I have to do new release with actual master...

Uup115, 2019-07-23:
@Pablo2048  I reverted back to lv_arduino (v2.0)  Yes, everything is working with both (latest master and lv_arduino).  Good to go!  

Thanks for everyone's help!

Pablo2048, 2019-07-23:
... so I'm closing this as solved...

closed 2019-07-23 by Pablo2048

embeddedt, 2019-07-23:
> I've already kicked myself several times. I'll go ahead and kick myself several more times for you, kisvegabor, and Pablo2048.

Now we'll never be able to add a disclaimer stating that no developers were harmed in the making of this product. :smile:

Seriously, don't worry about it. It's a pretty difficult issue to find when it surfaces.