lvgl/lv_arduino #19: "How to" question

shr1Khr, 2019-07-25:
Hey guys, 

So this is not an issue but more like how can I do question - 

**Objective**:
Im loading a screen and on a screen, I'm trying to show a text - 
"Please place your fingers now"
Then below that text show a 3 sec counter, via another text, and update just this text as - 
"3"
replaced with
"2"
replaced with 
"1"
replaced with 
"reading now"
Then I send the control to my function, it takes few seconds to finish and show preload on the screen in the meantime, then when my function finish, show results on the screen.

**Issue**
Now, I can't use a delay function for the counter or call any other function that takes few seconds to finish, since littlevgl doesn't load anything till delay function gets finished, or everything in a function gets finished and ticker function is called, so how do I do this in one function and not call the same function from loop and then "go to next step" kind of thing (since I'm using it this way now, calling the same function from loop, and increment the steps after every second with delay, which again is a bad idea as Im using delay). I'm using esp32, do you think I should move the ticker or littlevgl to second core, but then that would create another hassle of syncing everything or what should I do?

Pablo2048, 2019-07-25:
ESP32 Arduino core is based on RTOS so I'm using it this way:
In initialisation part I'm using
```C
  lvglMutex = xSemaphoreCreateMutex();
  xTaskCreate(gui_task, "lvgl", 10 * 1024, NULL, 5, NULL); // Start GUI task
```
gui_task method is defined like
```C
void gui_task(void *dev)
{

  while (1)
  {
    xSemaphoreTake(lvglMutex, portMAX_DELAY);
    lv_task_handler();
    xSemaphoreGive(lvglMutex);
    vTaskDelay(5);
    lv_tick_inc(5);
  }
  vTaskDelete(NULL);
}
```
And everywhere I need to be synced, I'm using ```xSemaphoreTake(lvglMutex, portMAX_DELAY);``` and ```xSemaphoreGive(lvglMutex);```
BTW: I suggest to use the forum for such kind of questions next time...

embeddedt, 2019-07-25:
I added the asynchronous call API to LittlevGL for cases exactly like this (inspired by JavaScript libraries): https://docs.littlevgl.com/en/html/overview/task.html#asynchronous-calls

It can fire a callback after 1 second without blocking execution.

And @Pablo2048 is correct...
----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2019-07-25 by embeddedt

shr1Khr, 2019-07-25:
Sure, thanks guys, Ill check out the forum.
The solution of @Pablo2048 is bit out of my skill set. Ill check it out in few days, perhaps will go with the solution given by @embeddedt

embeddedt, 2019-07-25:
@shr1Khr My solution is designed to work in any environment, regardless of threading capability, because it uses facilities already provided by LittlevGL.

What @Pablo2048 is suggesting is to basically move your countdown logic to a separate OS thread, **guarded by a mutex**. The other thread can sleep for 1 second but it won't affect LittlevGL, which continues running in its own thread.

That method will make for cleaner code but is more complex to set up at first and requires an RTOS.

shr1Khr, 2019-07-25:
I haven't used mutex logic yet, and thanks to your comment, I understand it's a better solution here.
Even though I did used multi-threading in Unity, I'm not too comfortable with it. As I will become more desperate, I'll perhaps use it. I suppose for now I'll stick with simpler solution. Thanks again guy.