lvgl/lv_arduino #28: Night Theme compile problem

momosh13, 2019-10-06:
Hi
I'm getting 

 'lv_theme_night_init' was not declared in this scope.

Please help me here, what i'm doing wrong? and i can fix it?
Thank you

embeddedt, 2019-10-06:
1. Did you enable it in `lv_conf.h`?
2. Are you including `lvgl.h`?

momosh13, 2019-10-07:
Thank you 
The "theme_night" works now, i have one more major problem.
The display, i'm trying to use this library for, is ILI9488, with Capacitive Touch / controller is FT6206,
and it's connect it with parallel connection, it's lot faster the SPI connection, but TFT-eSPI in parallel  connection, the  touch controller functions are not enabled, so i'm getting error when i'm trying to compile " ESP32_TFT_eSPI_Slider " ,
this is the error  'class TFT_eSPI' has no member named 'getTouch'
Please, help me whit this one too
Thank you again

embeddedt, 2019-10-07:
> the touch controller functions are not enabled, so i'm getting error when i'm trying to compile 

Please do not double post. You already asked about this in https://github.com/littlevgl/lvgl/issues/855.

closed 2019-10-07 by embeddedt

elgerg, 2019-10-07:
Sorry to potentially re-open but that issue "getTouch" is because you havent enabled touch in the bodmer tft_espi lib. 

Exact same query with bodmer (the lib creator):
https://github.com/Bodmer/TFT_eSPI/issues/117

See line:
//#define TOUCH_CS PIN_D2     // Chip select pin (T_CS) of touch screen

in User_Setup.h in the TFT_Espi lib folder.

Also you need this to calibrate the screen. Please see:

https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino

Any further issues with the TFT_Espi should really be directed to bodmer:

https://github.com/Bodmer/TFT_eSPI

Kind regards
Alex