lvgl/lv_arduino #29: V2.0.3 in library manager

elgerg, 2019-11-10:
Hi,

I went to update to the latest release but it's not there:

[image]

Is there something i need to do to be able to get the latest version though the lib manager?

Thanks
Alex

embeddedt, 2019-11-10:
My guess is that it needs to be published in some manner before it becomes visible in the Arduino library manager. But I'm not familiar with Arduino so I could be wrong.

I suppose you could temporarily replace the files in the interim if you need a quick solution.

Pablo2048, 2019-11-11:
Hi. I was out of hardware so I was unable to test the new version before release. I think that I can test it before weekend so please be patient...

closed 2019-12-09 by stale[bot]