lvgl/lv_arduino #31: Black screen

sonastyle, 2020-01-09:
Only getting Grey-black screen although touch is working properly.
I tested with custom design graphics and touch settings with core bodmer lib which are working perfectly.
Even if in the lvgl example  (TFT_ESPI_SLIDER) code I can use bodmer TFT_ESPI object and draw rect properly but lvgl code (slider, label) is not working any how.

Picture with default slider example
[image]

Picture with default slider example and blue fill screen and red rectangle draw using Tft_Espi object before and after lvgl code.
[image]

When I Touched the screen the following data is appeared also - 

Data x110
Data y103
Slider Pressed
SliderSliderSlider   Short clicked
Slider Clicked
Slider Released
Data x132
Data y96
Slider Pressed
SliderSliderData x129
Data y93
SliderSliderSlider     Short clicked
Slider Clicked
Slider Released

Please guide me if anything I'm missing.

Pablo2048, 2020-01-09:
So youre saying, that the LCD is working with just TFT_eSPI examples, but not with lv_arduino examples?

sonastyle, 2020-01-09:
> So youre saying, that the LCD is working with just TFT_eSPI examples, but not with lv_arduino examples?

- Yes, I have created several graphics like switch, slider by raw TFT_eSPI object which are also working with the example ...slider code but any lvgl function like label, slider is not working.

Display: 2.8" SPI ILI9341
IDE: Arduino ESP32 core 1.0.4

Pablo2048, 2020-01-09:
Are you using the same user_setup from TFT_eSPI examples with lv_arduino one? Can you please try the simplest lv_arduino example first (just label on the screen)? Which version on lv_arduino are you using?

sonastyle, 2020-01-09:
Same result with simple label example - No lvgl function works.

using - https://github.com/littlevgl/lv_arduino
- same user_setup from TFT_eSPI examples 

For ensuring that bodmer TFT_SPI and SPI conf is working I just add -  

tft.fillRect(20, 20, 100, 100, TFT_RED); //

 at the end of setup() and the red rectangle drawn perfectly but lvgl label does not work.

Please see the result
[image]

- Thanks for your co operation

Pablo2048, 2020-01-09:
So you are not using any version, installed by the Library manager? Can you please check, if older version of the library works with your LCD? You can get V2.0.3 from releases page.

sonastyle, 2020-01-09:
I replace the library with https://github.com/littlevgl/lv_arduino/releases/tag/2.0.3 but still no luck.

I'm attaching the code and lvgl_config.h

[code.txt](https://github.com/littlevgl/lv_arduino/files/4040022/code.txt)
[lv_conf.txt](https://github.com/littlevgl/lv_arduino/files/4040024/lv_conf.txt)

I'm using ESP32 VSPI bus and back light led directly connected with 3v3 vcc.

Pablo2048, 2020-01-09:
Well, it seems like I'm short of ideas what to do next... Can you please test it again with TFT_eSPI version 1.4.11? This is the version I've used.

sonastyle, 2020-01-10:
> Well, it seems like I'm short of ideas what to do next... Can you please test it again with TFT_eSPI version 1.4.11? This is the version I've used.

I updated TFT_eSPI to latest v1.4.20 (previously was 1.3.10) and its now working! Thanks a lot!

closed 2020-01-10 by sonastyle