lvgl/lv_arduino #32: Help Wanted

gsaikiran29, 2020-01-24:
hello, I am using roger's stm32_arduino core, sumotoy TFT_ILI93XX library.
I want to use This lv_arduino is this supported to that libraries.
I want to just replace the Bodmer/TFT_eSPI with sumotoy TFT_ILI93XX library. can any one help me to solve this.

gsaikiran29, 2020-01-24:
I changed This piece of code, This filled white color on screen but not printed the text.
please check it out
`
#include <lvgl.h>
#include "SPI.h"
#include "TFT_ILI93XX.h"

#define TFT_DC           PD9   // TFT display D/C pin
#define TFT_CS           PB12   // TFT display ChipSelect pin
#define TFT_RST          PD8    // TFT display ResetPIN
#define TFT_BACKLIGHT    PD10    // TFT display BackLight PIN
TFT_ILI93XX tft = TFT_ILI93XX(TFT_CS, TFT_DC, TFT_RST);
static lv_disp_buf_t disp_buf;
static lv_color_t buf[LV_HOR_RES_MAX * 10];

#if USE_LV_LOG != 0
/* Serial debugging */
void my_print(lv_log_level_t level, const char * file, uint32_t line, const char * dsc)
{

  Serial1.printf("%s@%d->%s\r\n", file, line, dsc);
  delay(100);
}
#endif

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint16_t c;

  SPI_TFT.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0)); /* Start new TFT transaction */
  tft.setAddrWindow_cont(area->x1, area->y1, (area->x2 - area->x1 + 1), (area->y2 - area->y1 + 1)); /* set the working window */
  for (int y = area->y1; y <= area->y2; y++) {
    for (int x = area->x1; x <= area->x2; x++) {
      c = color_p->full;
      tft.pushBlock(c, 1);
      color_p++;
    }
  }
  SPI_TFT.endTransaction(); /* terminate TFT transaction */
  lv_disp_flush_ready(disp); /* tell lvgl that flushing is done */
}

/* Interrupt driven periodic handler */
//static void lv_tick_handler(void)
//{
//
//  lv_tick_inc(LVGL_TICK_PERIOD);
//}

/* Reading input device (simulated encoder here) */
bool read_encoder(lv_indev_drv_t * indev, lv_indev_data_t * data)
{
  static int32_t last_diff = 0;
  int32_t diff = 0; /* Dummy - no movement */
  int btn_state = LV_INDEV_STATE_REL; /* Dummy - no press */

  data->enc_diff = diff - last_diff;;
  data->state = btn_state;

  last_diff = diff;

  return false;
}

void setup() {

  Serial1.begin(9600); /* prepare for possible serial debug */

  lv_init();

//#if USE_LV_LOG != 0
//  lv_log_register_print(my_print); /* register print function for debugging */
//#endif

  tft.begin(); /* TFT init */
  tft.setRotation(1); /* Landscape orientation */
  tft.useBacklight(TFT_BACKLIGHT);

  lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);

  /*Initialize the display*/
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = 320;
  disp_drv.ver_res = 240;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.buffer = &disp_buf;
  lv_disp_drv_register(&disp_drv);

  /*Initialize the touch pad*/
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_ENCODER;
  indev_drv.read_cb = read_encoder;
  lv_indev_drv_register(&indev_drv);

  /*Initialize the graphics library's tick*/
  //tick.attach_ms(LVGL_TICK_PERIOD, lv_tick_handler);

  /* Create simple label */
  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino! (V6.1)");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);
}

void loop() {

 lv_task_handler(); /* let the GUI do its work */
  delay(5);
}
`

Pablo2048, 2020-01-24:
Did you check (probably with some examples from sumotoy's library) that the hardware is working and the library is compatible with STM32 Core? Also please pay more attention with pasting code (use correct markdown formatting) and do not use generic topic names like "Help Wanted" next time...

Pablo2048, 2020-01-24:
After editing your comment and looking deeper into your code it seems like you commented out `lv_tick_handler` - this is wrong IMO...

gsaikiran29, 2020-01-24:
Thanks for the reply.
could you tell me first what it will do actually.?
should i add it to my hardware timer?

gsaikiran29, 2020-01-24:
> Did you check (probably with some examples from sumotoy's library) that the hardware is working and the library is compatible with STM32 Core? 

yes it works well. i am using this library from past 2 years

Pablo2048, 2020-01-24:
What `lv_tick_inc` does you can see in `lv_hal_tick.c` . IIRC without calling it some GUI elements get stuck. You have to attach calling lv_tick_inc for example to the timer interrupt (this is what Ticker object does). Also you have to comment out SPI_transaction things - in your sketch the SPI is not shared so it is not need...

gsaikiran29, 2020-01-24:
I added "lv_tick_handler" to a timer interrupt and removed SPI_transaction in this code
`
#include <lvgl.h>
#include "SPI.h"
#include "TFT_ILI93XX.h"

#define LED_RATE 1000000     // in microseconds; should give 0.5Hz toggles
HardwareTimer timer(1);      // Timer Object

#define TFT_DC           PD9   // TFT display D/C pin
#define TFT_CS           PB12   // TFT display ChipSelect pin
#define TFT_RST          PD8    // TFT display ResetPIN
#define TFT_BACKLIGHT    PD10    // TFT display BackLight PIN
TFT_ILI93XX tft = TFT_ILI93XX(TFT_CS, TFT_DC, TFT_RST);
static lv_disp_buf_t disp_buf;
static lv_color_t buf[LV_HOR_RES_MAX * 10];

#if USE_LV_LOG != 0
/* Serial debugging */
void my_print(lv_log_level_t level, const char * file, uint32_t line, const char * dsc)
{

  Serial1.printf("%s@%d->%s\r\n", file, line, dsc);
  delay(100);
}
#endif

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint16_t c;

  //SPI_TFT.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0)); /* Start new TFT transaction */
  tft.setAddrWindow_cont(area->x1, area->y1, (area->x2 - area->x1 + 1), (area->y2 - area->y1 + 1)); /* set the working window */
  for (int y = area->y1; y <= area->y2; y++) {
    for (int x = area->x1; x <= area->x2; x++) {
      c = color_p->full;
      tft.pushBlock(c, 1);
      color_p++;
    }
  }
  //SPI_TFT.endTransaction(); /* terminate TFT transaction */
  lv_disp_flush_ready(disp); /* tell lvgl that flushing is done */
}

/* Interrupt driven periodic handler */
static void lv_tick_handler(void)
{

  lv_tick_inc(20);
}

/* Reading input device (simulated encoder here) */
bool read_encoder(lv_indev_drv_t * indev, lv_indev_data_t * data)
{
  static int32_t last_diff = 0;
  int32_t diff = 0; /* Dummy - no movement */
  int btn_state = LV_INDEV_STATE_REL; /* Dummy - no press */

  data->enc_diff = diff - last_diff;;
  data->state = btn_state;

  last_diff = diff;

  return false;
}

void Timer_Init(void)
{
  timer.pause();
  timer.setPeriod(LED_RATE);                    // in microseconds    // Set up period
  timer.setChannel1Mode(TIMER_OUTPUT_COMPARE);  // Set up an interrupt on channel 1
  timer.setCompare(TIMER_CH1, 1);               // Interrupt 1 count after each update
  timer.attachCompare1Interrupt(lv_tick_handler);
  timer.resume();                               // Start the timer counting
}

void setup() {

  Serial1.begin(9600); /* prepare for possible serial debug */

  lv_init();

//#if USE_LV_LOG != 0
//  lv_log_register_print(my_print); /* register print function for debugging */
//#endif

  tft.begin(); /* TFT init */
  tft.setRotation(1); /* Landscape orientation */
  tft.useBacklight(TFT_BACKLIGHT);

  lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);

  /*Initialize the display*/
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = 320;
  disp_drv.ver_res = 240;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.buffer = &disp_buf;
  lv_disp_drv_register(&disp_drv);

//  /*Initialize the touch pad*/
//  lv_indev_drv_t indev_drv;
//  lv_indev_drv_init(&indev_drv);
//  indev_drv.type = LV_INDEV_TYPE_ENCODER;
//  indev_drv.read_cb = read_encoder;
//  lv_indev_drv_register(&indev_drv);

  /*Initialize the graphics library's tick*/
  //tick.attach_ms(LVGL_TICK_PERIOD, lv_tick_handler);
  Timer_Init();

  /* Create simple label */
  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino! (V6.1)");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);
}

void loop() {

  lv_task_handler(); /* let the GUI do its work */
  delay(5);
}
`

gsaikiran29, 2020-01-24:
also you ca check my core libraries
[https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries](url)
[https://github.com/sumotoy/TFT_ILI93XX](url)

Pablo2048, 2020-01-24:
Bad formatting again :-( ... Actually I don't know if it is correct - I don't use STM32 Arduino core. Why you comment out the dummy encoder input device? Did you try to enable serial port debugging?

gsaikiran29, 2020-01-24:
I commented that because my display non touch screen display. I thought Touch intializations may not work.

embeddedt, 2020-01-24:
Have you tried enabling logging? What output do you see? Remember that you can use Markdown code blocks to format your code and make it easier to read (` ```c ` and ` ``` `).

Pablo2048, 2020-01-24:
There are no touch library in your sketch - if you take a look into the code you can see that it is just dummy input driver just to make lvgl internals happy. Please enable logging...

gsaikiran29, 2020-01-24:
I have tried to enable logging it's giving compilation error

> C:\Users\AMP_SAI\Desktop\ESP32_TFT_eSPI\ESP32_TFT_eSPI.ino: In function 'void setup()':
>ESP32_TFT_eSPI:83:33: error: 'lv_log_register_print' was not declared in this scope
>   lv_log_register_print(my_print); /* register print function for debugging */
>                                 ^

Pablo2048, 2020-01-24:
Did you modified in `lv_conf.h` values of `LV_USE_LOG` and `LV_LOG_LEVEL`?

embeddedt, 2020-01-24:
I believe it should be `lv_log_register_print_cb`.

gsaikiran29, 2020-01-24:
no I have changed that in my code only. now I have changed in lv_conf.h it is compiled successfully. but its not printing any data on my serial port

embeddedt, 2020-01-24:
Do you have access to a debugger? Stepping through the code might be the fastest way of figuring out what the issue is, instead of us having to guess potential problems. Can you try that? AFAIK `lv_init` should print something to the log, so stepping through that function to find out why nothing prints is probably a good idea.

Pablo2048, 2020-01-24:
Did you init the serial port first?

gsaikiran29, 2020-01-24:
Yes in my setup() function I initialized
```c
Serial1.begin(9600); /* prepare for possible serial debug */
```
and this is the log print function
```c#if USE_LV_LOG != 0
/* Serial debugging */
void my_print(lv_log_level_t level, const char * file, uint32_t line, const char * dsc)
{

  Serial1.println(String(file)+"@"+String(line)+"->"+String(dsc));
  delay(100);
}
#endif
```

Pablo2048, 2020-01-24:
For 9066Bd the 100ms delay seems to be too short, but anyway it has to send out something. What is your `LV_LOG_LEVEL`?

gsaikiran29, 2020-01-24:
#define LV_LOG_LEVEL_WARN 2

Pablo2048, 2020-01-24:
:-( can you please try it with LV_LOG_LEVEL_TRACE ? WARN shows only warnings (as described in lv_conf.h)

gsaikiran29, 2020-01-24:
I changed that too no use:-(
```c
#define LV_USE_LOG      1
#if LV_USE_LOG
/* How important log should be added:
 * LV_LOG_LEVEL_TRACE       A lot of logs to give detailed information
 * LV_LOG_LEVEL_INFO        Log important events
 * LV_LOG_LEVEL_WARN        Log if something unwanted happened but didn't cause a problem
 * LV_LOG_LEVEL_ERROR       Only critical issue, when the system may fail
 * LV_LOG_LEVEL_NONE        Do not log anything
 */
#  define LV_LOG_LEVEL    LV_LOG_LEVEL_TRACE
```

Pablo2048, 2020-01-24:
So you have to figure out why this line https://github.com/littlevgl/lv_arduino/blob/0e11c3a0fe95b2e58350d72178697d40e7a2cd1d/src/src/lv_core/lv_obj.c#L87 doesn't show on serial port first...

gsaikiran29, 2020-01-24:
in lv_obj.c file i have tried to debug by adding this
```c
void lv_init(void)
{
	Serial1.println("init started");
```
but its giving error Serial1 not declared in this scope

Pablo2048, 2020-01-24:
Well, its obvious - you try to call c++ object from c source code... It seems like you are not very experenced so I have to ask - did you reset the board after connecting the serial terminal? You have to be pretty sure that the serial terminal is connected BEFORE `lv_init()` is called...

gsaikiran29, 2020-01-25:
> did you reset the board after connecting the serial terminal?
yes

> You have to be pretty sure that the serial terminal is connected BEFORE lv_init() is called...
yes before only serial connected

Pablo2048, 2020-01-25:
You can try to put `Serial1.println("LVGL test");` before lv_init() is called and watch, if it prints out the text...

Pablo2048, 2020-01-31:
@gsaikiran29 any news on this, or can I close this issue?

gsaikiran29, 2020-01-31:
No Im trying to solve this one..
before lv_init() also not printing anything

Pablo2048, 2020-02-08:
Closing due to lack of activity. If you think, that there is something wrong in the library, please open a new issue and post your investigation results as to why you think so.

closed 2020-02-08 by Pablo2048