lvgl/lv_arduino #33: Lib updates not appearing in Arduino IDE library manager

Stego72, 2020-03-08:
Hi 

The latest versions do not appear in the Arduino IDE Library Manager - only versions up to 2.0.1 appear. Hence update functionality is broken.

BR

Stego72

Pablo2048, 2020-03-10:
Hi,
thanks for spotting this. I've made new request (because of changed repository name) here https://github.com/arduino/Arduino/issues/9850 so we have to waitnow...

Pablo2048, 2020-03-27:
Well, my request was completed 7 days ago, but I still don't see the new version in my Arduino IDE. Can you @Stego72 please check it?

Stego72, 2020-03-30:
Hi - Nope 

Now shows versions up to 2.0.3 (2.0.1 before) but none of the versions after that

BR

Stego72

Pablo2048, 2020-04-24:
Done - there is v2.1.4 in Arduino IDE Library manager present so I'm closing this as solved.

closed 2020-04-24 by Pablo2048

kisvegabor, 2020-04-24:
Thank you Pablo!