lvgl/lv_arduino #34: Use millis() as custom tick

kisvegabor, 2020-03-20:
Hi Pablo,

What would you think about setting `millis()` [lv_conf.h](https://github.com/littlevgl/lv_arduino/blob/master/lv_conf.h#L202-L206) as a custom tick function?

This way `Ticker` wouldn't be required.

Pablo2048, 2020-03-24:
Hi Gabor,
sorry for late reply. I thought that using Ticker lower the CPU load. Unfortunately Ticker is only ESP8266/ESP32 specific so I take a look and change the example to use `millis()`...

kisvegabor, 2020-03-24:
No problem! :) 

> Unfortunately Ticker is only ESP8266/ESP32 specific so I take a look and change the example to use millis()...

In this case, a more portable solution would be really better. Thank you in advance!

Pablo2048, 2020-04-15:
I'm working on it - it is done actually in my git repo, but I'm waiting for LCD module to arrive...

kisvegabor, 2020-04-15:
Great, thank you for the update!

closed 2020-05-13 by stale[bot]