lvgl/lv_arduino #35: Display distorted with STM32F103 and ILI9341

HamidSaffari, 2020-03-25:
Hi and salute to the creators.
I tried to run this with STM32F103 and ILI9341 using https://github.com/stm32duino/Arduino_Core_STM32
core and your ESP32_TFT_eSPI example works fine to showing hello-world but when I used ESP32_TFT_eSPI_Slider example I have got distorted image like the picture:

[image]

Pablo2048, 2020-03-25:
Hi,
did you modified the screen resolution as mentioned here https://github.com/littlevgl/lv_arduino/tree/master/examples/ESP32_TFT_eSPI_Slider#example-for-lv_arduino-using-a-slider ?

HamidSaffari, 2020-03-25:
Oh man, I forget to modified resolution in ESP32_TFT_eSPI_Slider.ino and only did that in lv_conf.h. Now it Works. Thanks.
I'm trying to make an example with STM32 and as soon as it's finish I will make a pull-request.
also I find out that compile result when changing memory wrappers by: 
#define LV_MEM_CUSTOM   1 //compile result => FLASH: 107720 bytes, RAM: 13444 bytes 
vs
#define LV_MEM_CUSTOM   0 //compile result => FLASH: 107904 bytes, RAM: 46216 bytes 

saving  32777 bytes of RAM  which most of them for LV_MEM_SIZE = 32K.
now is there any instability or slowness in performance if we use custom wrappers?

Pablo2048, 2020-03-25:
Great - so this issue is solved. PR with your example for STM32 is welcome :-) . About your memory question - better ask directly in master lvgl repository.

closed 2020-03-25 by Pablo2048

kisvegabor, 2020-03-25:
There are 2 purposes of the built-in memory functions:
1. Platform independent behavior
2. Memory monitoring feature (free/used memory, fragmentation, etc)

So other `malloc/free` implementations can work equally well if memory monitoring is not important for you.

(The forum would be really better a place for these general questions.)

HamidSaffari, 2020-03-25:
Thanks Gabor & all the contributors for making this great and complicated library. Yes multi-platform support makes it more complicated, you already surpassed non-free GUI makers software like Embedded-Wizard, SEGGER-emWin, TouchGFX in many cases and I hope by finishing your drag&drop app [lv_gui_builder] you beat them completely.
So keep it coming.

kisvegabor, 2020-03-27:
Thank you Hamid! :slightly_smiling_face: 
If you have experience with other GUI libraries I'd love to read about it a Forum post.