lvgl/lv_arduino #36: Can't compile project

PsyonicOne, 2020-03-31:
Sorry if this is the wrong place for this... I couldn't find anywhere else...

I'm using Atom on Linux (v 1.44.0) with PlatformIO-ide (v 2.6.0)
I'm using the standard method of installing pio libs inside Atom.
Without including lvgl.h I can produce a perfectly working screen.

When I
`#include <lvgl.h>`
in my project and build it spits out A LOT of errors!

I had the same thing when compiling the lvgl library from source, but eventually realised that all I needed to do was change the
`#if 0 /*Set it to "1" to enable content*/`
to 1 (which mind you is not mentioned in the "Quick Start" docs, took me ages to realise this... maybe I was looking in the wrong places...)

But I can't find any issue like that in your Arduino library...

What am I missing??

Pablo2048, 2020-03-31:
With so small and abstract informations I'm unable to help. "A LOT of errors" - example? At least the first one. You use Atom with PIO and the Arduino framework?

PsyonicOne, 2020-03-31:
Sorry, late night.
Correct, Atom with PIO on Fedora 31.
I had the same thing trying to build before I knew about setting #if 1 in lv.conf.h
I've attached the build log. Thanks for your time.
[build.log](https://github.com/littlevgl/lv_arduino/files/4411993/build.log)

Pablo2048, 2020-04-01:
This is a mess :-( . Please remove unused libraries from your project first and try to clean other errors also (`/bin/sh: module: line 1:syntax error: unexpected end of file`...). Also please read this https://github.com/littlevgl/lvgl#supported-devices - lvgl is written for at least 16 bit CPU and your ATMEGA2560 is 8 bit platform only with 8kB RAM (also see RAM section in supported devices) - probably this explain every `-Wnarrowing` warnings and out of memory eror in `work_mem_int`. I'm not sure that this library will work in this memory constrained device.

PsyonicOne, 2020-04-01:
Ah, thanks.
lol, yeah, it IS a mess! That's why I'm asking for help!

There are only 3 libraries in the project
```
#include <TouchScreen.h>
#include <lvgl.h>
#include <MCUFRIEND_kbv.h>
```

I always get loads of `/bin/sh: module: line 1:syntax error: unexpected end of file` compiling in the PIO IDE, even with a fresh project with literally nothing in it, just a single main.ino with
```
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
but correct code compiles successfully, uploads and runs just fine, I just ignore it as I have no idea why it's happening and it does not seem to matter.

Anyway, thanks for looking at it for me, I'll try it some day with a better controller. :-}

Pablo2048, 2020-04-01:
You have something wrong in your Linux configuration - see https://github.com/AtomLinter/linter-flake8/issues/165 . Anyway I'm closing this...

closed 2020-04-01 by Pablo2048