lvgl/lv_arduino #39: littlevgl  'lv_theme_night_init' was not declared in this scope compile error

williamcustinne, 2020-04-07:
I'm getting 'lv_theme_night_init' was not declared in this scope. 
theme is enabled lv_conf.h
lvgl.h is included
lv_conf.h in same directory as lvgl.h
I have no idea what more can be wrong can somebody help ?

embeddedt, 2020-04-08:
Are you sure that that `lv_conf.h` file is being used? Adding an `#error` statement is a good way to check.

williamcustinne, 2020-04-08:
yes lv_conf.h is correct and in the right path.

williamcustinne, 2020-04-08:
After deleting the lib completely and reinstalling it with Arduino lib manager, all theexamples from examples master are working. The problem persist ONLY with the example ESP32_TFT_eSPI_SLIDER. It must have to do something with the relative path to lv_conf.h. lv_config.h in the root of lv_arduino seems to work for the other examples.

timmcarson, 2020-04-08:
I know you said the theme was enabled but I had this same error until I turned the correct theme on in lv_conf.h  halfway down in the file you should see this.  is there a one next to the night them?

[image]

williamcustinne, 2020-04-08:
the night theme is enabled. Software for the IdeasNProjects-master have also a scetch with a slider using the night theme. This one works perfect. The code is setup into one file for example. The example that comes with lv_arduino -->  ESP_TFT_eSPI_SLIDER has several tabss (files) and does not work and produces the error. Even saving the example to another location (relative paths ...) does not resolve the problem.

timmcarson, 2020-04-08:
I am using the Arduino framework inside PlatformIO so what I did may not help much.

I combined all the tabs into one file.  (Arduino.h should not be needed if using arduino IDE)
[main.txt](https://github.com/littlevgl/lv_arduino/files/4451755/main.txt)

in the platformio.ini file, I defined LV_CONF_INCLUDE_SIMPLE  Not sure how to do this in arduino IDE maybe just `#define LV_CONF_INCLUDE_SIMPLE`

I also had an issue with "lv_conf_checker.h"  so at the bottom of lv_conf.h file I removed the path
so the include looks like this
`#include "lv_conf_checker.h"`

JimDrewGH, 2020-05-05:
I can also confirm the same error with a ESP32.  The other included example (ESP32_TFT_eSPI) works fine.  I am using the Arduino IDE only (not platform IO).

Pablo2048, 2020-05-05:
I did try to reproduce the problem, unfortunately I can't. I've installed Arduino IDE  v 1.8.12 in portable manner, OS is Windows 7 Pro, then I installed `lv_arduino` in version 2.1.4 and `TFT_eSPI` v 2.2.0 via Library manager. Then I installed ESP32 Core v 1.0.4. I've select `ESP32 Wrover module` as an target, manually edit `lv_conf.h` file in `lv_arduino` library directory to enable `#define LV_USE_THEME_NIGHT      1   /*Dark elegant theme*/ ` and opened `ESP32_TFT_eSPI_Slider` demo example. After commenting out all touch references (I don't want to modify TFT_eSPI configuration), the compilation vent just fine.

Pablo2048, 2020-05-24:
No response, so I'm closing this as an user error...

closed 2020-05-24 by Pablo2048

szybkijanek, 2020-06-11:
Hi, I've same problem, 'lv_theme_night_init' not declared, i do same instructions like:

> I did try to reproduce the problem, unfortunately I can't. I've installed Arduino IDE v 1.8.12 in portable manner, OS is Windows 7 Pro, then I installed `lv_arduino` in version 2.1.4 and `TFT_eSPI` [...]

but stil have error because file don't contain these lines:
/*================

THEME USAGE
================/
#define LV_THEME_LIVE_UPDATE 1 /1: Allow theme switching at run time. Uses 8..10 kB of RAM/
#define LV_USE_THEME_TEMPL 0 /Just for test/
#define LV_USE_THEME_DEFAULT 0 /Built mainly from the built-in styles. Consumes very few RAM/
#define LV_USE_THEME_ALIEN 0 /Dark futuristic theme/
#define LV_USE_THEME_NIGHT 1 /Dark elegant theme/
#define LV_USE_THEME_MONO 0 /Mono color theme for monochrome displays/
#define LV_USE_THEME_MATERIAL 0 /Flat theme with bold colors and light shadows/
#define LV_USE_THEME_ZEN 0 /*Peaceful, mainly light theme */
#define LV_USE_THEME_NEMO 0 /Water-like theme based on the movie "Finding Nemo"/

when I added it, it's no effect. Lib isn't compatible with previous version so i must change program? Jesus, next "dev" lib ...