lvgl/lv_arduino #40: When i use lv_obj_del() my ESP32 reboots

albertomercurio, 2020-04-11:
Hello,
i create an object list:
```
      WiFi.mode(WIFI_STA);
      WiFi.disconnect();
      delay(50);

      Serial.println("Scan started");
      int n = WiFi.scanNetworks();
      Serial.println("Scan done");
      if (n == 0) {
        Serial.println("No networks found");
      } else{
        String ssid[n];
        String rssi[n];
        for (int i = 0; i < n; ++i) {
          Serial.print(WiFi.SSID(i));
          Serial.print(" (");
          Serial.print(WiFi.RSSI(i));
          Serial.println(")");

          ssid[i] = WiFi.SSID(i);
          rssi[i] = WiFi.RSSI(i);
          delay(10);
        }

        wifi_lst = lv_list_create(lv_scr_act(), NULL);
        lv_obj_set_size(wifi_lst, 300, 180);
        lv_obj_align(wifi_lst, NULL, LV_ALIGN_CENTER, 0, 0);

        byte buffer_len = 100;
        for(int i = 0; i<n; i++) {
          char ssid_char[buffer_len];
          ssid[i].toCharArray(ssid_char, buffer_len);
          lv_obj_t *btn_tmp = lv_list_add_btn(wifi_lst, LV_SYMBOL_WIFI, ssid_char);
          lv_obj_set_event_cb(btn_tmp, ssid_selected);
        }
      }
```
where my event function is:
```
void ssid_selected(lv_obj_t * obj, lv_event_t event){
  lv_obj_del(wifi_lst);
}
```
But when i click the item the ESP32 reboots automatically.

Pablo2048, 2020-04-12:
Please provide MCVE, or we are unable to help.

albertomercurio, 2020-04-12:
Main.ino:
```
#include <lvgl.h>
#include <Ticker.h>
#include <TFT_eSPI.h>
#include <WiFi.h>

#define LVGL_TICK_PERIOD 60

Ticker tick; /* timer for interrupt handler */
TFT_eSPI tft = TFT_eSPI(); /* TFT instance */
static lv_disp_buf_t disp_buf;
static lv_color_t buf[LV_HOR_RES_MAX * 10];

lv_obj_t * slider_label;
lv_obj_t *wifi_lst;
int screenWidth = 480;
int screenHeight = 320;
  
#if USE_LV_LOG != 0
/* Serial debugging */
void my_print(lv_log_level_t level, const char * file, uint32_t line, const char * dsc)
{

  Serial.printf("%s@%d->%s\r\n", file, line, dsc);
  delay(100);
}
#endif

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint16_t c;

  tft.startWrite(); /* Start new TFT transaction */
  tft.setAddrWindow(area->x1, area->y1, (area->x2 - area->x1 + 1), (area->y2 - area->y1 + 1)); /* set the working window */
  for (int y = area->y1; y <= area->y2; y++) {
    for (int x = area->x1; x <= area->x2; x++) {
      c = color_p->full;
      tft.writeColor(c, 1);
      color_p++;
    }
  }
  tft.endWrite(); /* terminate TFT transaction */
  lv_disp_flush_ready(disp); /* tell lvgl that flushing is done */
}

bool my_touchpad_read(lv_indev_drv_t * indev_driver, lv_indev_data_t * data)
{
    uint16_t touchX, touchY;

    bool touched = tft.getTouch(&touchX, &touchY, 600);

    if(!touched)
    {
      return false;
    }

    if(touchX>screenWidth || touchY > screenHeight)
    {
      Serial.println("Y or y outside of expected parameters..");
      Serial.print("y:");
      Serial.print(touchX);
      Serial.print(" x:");
      Serial.print(touchY);
    }
    else
    {

      data->state = touched ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL; 
  
      /*Save the state and save the pressed coordinate*/
      //if(data->state == LV_INDEV_STATE_PR) touchpad_get_xy(&last_x, &last_y);
     
      /*Set the coordinates (if released use the last pressed coordinates)*/
      data->point.x = touchX;
      data->point.y = touchY;
  
      Serial.print("Data x");
      Serial.println(touchX);
      
      Serial.print("Data y");
      Serial.println(touchY);

    }

    return false; /*Return `false` because we are not buffering and no more data to read*/
}

/* Interrupt driven periodic handler */
static void lv_tick_handler(void)
{

  lv_tick_inc(LVGL_TICK_PERIOD);
}
```

Setup.ino:
```
void setup() {

  ledcSetup(10, 5000/*freq*/, 10 /*resolution*/);
  ledcAttachPin(32, 10);
  analogReadResolution(10);
  ledcWrite(10,768);

  Serial.begin(115200); /* prepare for possible serial debug */

  lv_init();

  #if USE_LV_LOG != 0
    lv_log_register_print(my_print); /* register print function for debugging */
  #endif

  tft.begin(); /* TFT init */
  tft.setRotation(1);

  uint16_t calData[5] = { 263, 3642, 298, 3534, 3 };
  tft.setTouch(calData);

  lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);

  /*Initialize the display*/
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = screenWidth;
  disp_drv.ver_res = screenHeight;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.buffer = &disp_buf;
  lv_disp_drv_register(&disp_drv);

  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);             /*Descriptor of a input device driver*/
  indev_drv.type = LV_INDEV_TYPE_POINTER;    /*Touch pad is a pointer-like device*/
  indev_drv.read_cb = my_touchpad_read;      /*Set your driver function*/
  lv_indev_drv_register(&indev_drv);         /*Finally register the driver*/

  /*Initialize the touch pad*/
  //lv_indev_drv_t indev_drv;
  //lv_indev_drv_init(&indev_drv);
  //indev_drv.type = LV_INDEV_TYPE_ENCODER;
  //indev_drv.read_cb = read_encoder;
  //lv_indev_drv_register(&indev_drv);

  /*Initialize the graphics library's tick*/
  tick.attach_ms(LVGL_TICK_PERIOD, lv_tick_handler);

  //Set the theme..
  lv_theme_t * th = lv_theme_night_init(210, NULL);     //Set a HUE value and a Font for the Night Theme
  lv_theme_set_current(th);

  lv_obj_t * scr = lv_cont_create(NULL, NULL);
  lv_disp_load_scr(scr);

  //lv_obj_t * tv = lv_tabview_create(scr, NULL);
  //lv_obj_set_size(tv, lv_disp_get_hor_res(NULL), lv_disp_get_ver_res(NULL));

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(10);

  int n = WiFi.scanNetworks();
  if (n == 0) {

  } else{
    String ssid[n];
    String rssi[n];
    for (int i = 0; i < n; ++i) {
      ssid[i] = WiFi.SSID(i);
      rssi[i] = WiFi.RSSI(i);
      delay(10);
    }

    wifi_lst = lv_list_create(lv_scr_act(), NULL);
    lv_obj_set_size(wifi_lst, 300, 180);
    lv_obj_align(wifi_lst, NULL, LV_ALIGN_CENTER, 0, 0);

    byte buffer_len = 100;
    for(int i = 0; i<n; i++) {
      char ssid_char[buffer_len];
      (ssid[i] + "  (" + rssi[i] + " dB)").toCharArray(ssid_char, buffer_len);
      lv_obj_t *btn_tmp = lv_list_add_btn(wifi_lst, LV_SYMBOL_WIFI, ssid_char);
      lv_obj_set_event_cb(btn_tmp, ssid_selected);
    }
  }
}
```

Loop.ino:
```
void loop() {

  lv_task_handler(); /* let the GUI do its work */
  delay(5);
}
```

Callbacks.ino:
```
void ssid_selected(lv_obj_t * obj, lv_event_t event){
  lv_obj_del(wifi_lst);
}
```

albertomercurio, 2020-04-12:
Solved using the following callback function:
```
void ssid_selected(lv_obj_t * obj, lv_event_t event, String ssid){
  if(event == LV_EVENT_SHORT_CLICKED){
    lv_btn_set_state(wifi_btn, LV_BTN_STATE_TGL_REL);
    lv_obj_del(wifi_lst);
  }
}
```

closed 2020-04-12 by albertomercurio