lvgl/lv_arduino #42: Arduino get started page

kisvegabor, 2020-05-07:
Hi @Pablo2048,

In the docs of v7 there will be sites for manufacturers and platforms. 
For Arduino it is here: https://docs.littlevgl.com/v7/en/html/get-started/arduino.html

It's more or less like a README. A few lines about Arduino, some recommended boards, how to get started, some platform-specific info, a super simple example, etc.

I'd like to ask if you have time and energy to write about these things.

Pablo2048, 2020-05-07:
Hi @kisvegabor ,
of course I will try to do my best and take a look (probably at the weekend).
BTW: I got message from my post office - I hope that it will be the missing TFT display s I can check the `millis()` thing also...

kisvegabor, 2020-05-08:
>  of course I will try to do my best and take a look (probably at the weekend).

Awesome, thank you!

> BTW: I got message from my post office - I hope that it will be the missing TFT display s I can check the millis() thing also...

Great! Delivery is unbelievably slow nowadays...

closed 2020-06-05 by stale[bot]