lvgl/lv_arduino #43: V7  Update

momosh13, 2020-05-19:
Please update lv_arduino to V7.01, when possible.
Thank you.

Pablo2048, 2020-05-25:
I'm waiting for final 7.0.1 release - 7.0.1-dev version is prepared in my repository. Only the first (and siplmest) example is checked because of my limited hardware availability.

kisvegabor, 2020-05-25:
Okay. The plan is to release at least a bugfix version every 2 weeks. So v7.0.1 is scheduled to 1 Jun.

kisvegabor, 2020-06-05:
@Pablo2048 
What do you have to do to release a new Arduino version? If it's possible I can update lvgl's [release script](https://github.com/lvgl/lvgl/blob/master/scripts/release.py) to apply some changes in this repo too.

Pablo2048, 2020-06-05:
I have to get the new version, check configuration if it's still valid, check at least the simplest example if it still works and if so I have to modify `library.json` and `library.properties' with new numbers. As I wrote here https://forum.lvgl.io/t/arduino-questions/2405/7?u=pablo2048 I hope that I'll do all this until end of this week.

kisvegabor, 2020-06-05:
Would you agree to automatically update library.json on patch releases? 
They don't add new features or config options just fixes some bugs.

Pablo2048, 2020-06-05:
Okay, no problem with that...

kisvegabor, 2020-06-05:
Great! I'll update the script then.

Pablo2048, 2020-06-07:
New PR has been created. Can anybody test it (especially with the STM32 example) or should I merge it?

closed 2020-06-08 by Pablo2048