lvgl/lv_arduino #50: Version 3 - Slider example

Stego72, 2020-06-08:
Hi

Has the slider example been modified to work with the new version? 

I'm having some issues relating to the "old" night theme - specifically:

`lv_theme_t * th = lv_theme_night_init(210, NULL); `

 - I notice in the SRC/LV_THEMES that this is now handled in a different way and that it no longer can be enabled in lv_conf.h

BR

Pablo2048, 2020-06-08:
Hi,
no, it wasn't - as I wrote in the release comment - I have only hardware for testing the simplest example so I'm unable to test slider/STM32 ones.

Pablo2048, 2020-06-08:
Perhaps @elgerg can test it and update...

Stego72, 2020-06-08:
I can also test/update - just need an intro to how the "themes" now work

embeddedt, 2020-06-08:
The [LVGL documentation](https://docs.lvgl.io/v7/en/html/overview/style.html#themes) is a good place to start for information about styles and themes.

elgerg, 2020-06-09:
> Perhaps @elgerg can test it and update...

I can give the slider example a go tomorrow/Thursday on the esp 32

Stego72, 2020-06-10:
I have updated the ESP32 slider example and created a pull request. It now works for 3.0.0 

Main changes: Removed references to "ticker" + updated the theme part to adhere to the new system.

(I'm a massive noob at github so if I have done something wrongly let me know)

Pablo2048, 2020-06-11:
So the slider example is updated - i have two options:

  * release new version (3.0.1)
  * or wait till someone check/update the STM32

suggestions anyone?

kisvegabor, 2020-06-11:
v7.0.3 will be released next Tuesday. (16.06)
I suggest waiting for it at least.

Pablo2048, 2020-06-11:
Thank you for the information, ok - I'll wait then.

Pablo2048, 2020-06-20:
Version 3.0.1, based on LVGL 7.0.2 was released. ESP32 slider example is corrected. No progress to STM32 unfortunately...

kisvegabor, 2020-06-22:
Pablo, @C47d [had the idea](https://forum.lvgl.io/t/rfc-esp32-port-repo-improvement-suggestions/2485) to join the ESP repo directly into to LVGL repo. 

I wonder if it would be possible with the Arduino project. It would be great to have Arduino examples in this repo, but releasing the Arduino lib directly from LVGL made things easier. 

We already have a [library.json](https://github.com/lvgl/lvgl/blob/master/library.json) file in lvgl for PlatformIO. At first look, it has some common points with the Arduino library.json.

I'm not sure what other files are required, or what directory structure needs to be followed. 

What do you think about it?

Pablo2048, 2020-06-23:
There are few things in Arduino libraries:
1. library has to follow Library Format Rules ( https://arduino.github.io/arduino-cli/library-specification/ ) - library.properties file and also directory names (/src, /examples, /extras).
2. unfortunately the original Arduino IDE can not handle user configurations of libraries very well - best way so far how to handle user configurations is decribed here https://github.com/Bodmer/TFT_eSPI/issues/661 - this can mix up things with PIO, or confuse users.
I personally not use Arduino IDE for years, so I'm okay with merging lv_arduino to one PIO/Arduino library, but we have to thing first, that oridnary Arduino IDE users are not very experienced...

kisvegabor, 2020-06-23:
According to [this example](https://arduino.github.io/arduino-cli/library-specification/#a-complete-example) we are not so far. Except `lvgl.h` is in `lvgl` and not `lvgl/src` and we don't have examples and extras folder but they not mandatory I suppose.

I could compile a library by doing this:
1. copy the original lvgl repo into Arduino/libraries
2. add library.properties from here into the lvgl folder
3. copy lvgl.h into the src folder
4. add lv_conf.h in the libraries folder.

To only the thing the user needs to do is creating lv_conf.h, however, he needs to use this file in the current case too. So expecting to copy `lv_conf_template.h` it seems like a doable thing. 

Anyway, what do you think, is it a good direction to make the core lvgl as an Arduino lib and keep this repo for examples?

Pablo2048, 2020-06-23:
Unfortunately no - I don't think that it is a good direction. Examples from every library installed are automatically "installed" with the library so they pops-up in Arduino-IDE File->Examples. If we goes like You suggested, then user need to install lvgl library as core, and another dummy library with examples.

kisvegabor, 2020-06-23:
What if we do the same with `lv_examples` too? There can be a simple ino file there in whichusers can call `lv_ex_...` or `lv_demo_..` functions.

Pablo2048, 2020-06-23:
So the user have to install `lvgl` first and then `lv_examples`? If so, then then lv_examples need structure refactoring - every single example need their own directory inside `/examples`, directory name must be the same as main `.ino` file. Im afraid that it goes to be overcomplicated for both maintainer of examples and for ordinary Arduino IDE user...

kisvegabor, 2020-06-23:
I meant creating only one example ino like this:
```c

#include "lvgl.h"
#include "lv_examples.h"

void setup() 
{
    lv_init();

    //Hardware init...
  
    lv_ex_btn_1(); //Or call other functions here to try ther examples
}

void loop ()
{
}
```

Pablo2048, 2020-06-23:
This approach need `lv_examples` structure like this:
```
/examples - this is mandatory 'examples' base
  + /ard_example - directory with one example
        + ard_example.ino - mandatory 'main' example file - same name as the directory
        + directory structure of all .c files
```
because the Arduino IDE copies only one directory on example invocation - the `ard_example`. I'm afraid that the IDE will find any of the original files. Maybe when installed separately as an library it will work, but I'm not sure...

kisvegabor, 2020-06-23:
I've tried it and it works.

Here is the hacked `lv_examples` library. 
[lv_examples.zip](https://github.com/lvgl/lv_arduino/files/4818745/lv_examples.zip)

Pablo2048, 2020-06-23:
:D it compile, but it won't work - for full Arduino example we need to add TFT driver, input driver, ... How to achieve this? The Arduino guys expect that the examples "just work" out of the box (even in lv_arduino if you have "default" ILI9341 hardware, connected to the ESP32, the example work immediately).

kisvegabor, 2020-06-23:
Sorry, if I still saying non sense things. Try my best to understand it :sweat_smile: 

I copy-pasted your ESP32_TFT_eSPI.ino, added `#include <lv_example.h>` and called `lv_ex_btn_1()` and it compiles.  [main.ino.zip](https://github.com/lvgl/lv_arduino/files/4819097/main.ino.zip)

Not sure if it works, but honestly I have no idea from where TFT_eSPI know the type of the display controller, port settings etc.

Pablo2048, 2020-06-23:
Ah, ok. So when we create Arduino examples such way, we can just add some comments that the Arduino ESP32 examples depend on TFT_eSPI library and user has to check its configuration as described here https://github.com/lvgl/lv_arduino#example . Ok, so I'll have to test all examples given on my ESP32 board and prepare "root" .ino files... :-)

kisvegabor, 2020-06-23:
Checking the link now I see where the settings of TFT_eSPI are. :slightly_smiling_face: 

You don't have to test all of them. If one works (e.g. `lv_ex_btn_1()`) the rest should work too. 
My vision is to have only 1 "root.ino" (with TFT_eSPI?) where the user can call any `lv_ex_...` functions.

It makes the maintenance of the examples even easier. All the examples are "normal" C files and regularly tested.

Pablo2048, 2020-06-23:
Hmm, but it's against Arduino example philosophy - one big/universal example, which has to be edited from user, but as I've written - I'm not using Arduino IDE so I'm fine with this...

kisvegabor, 2020-06-24:
Yes, it's kind of compromise. It still makes it easier to access all the examples and demos. 

I'll open an issue for it in the lvgl repo to discuss it with others too.

closed 2020-07-25 by stale[bot]