lvgl/lv_arduino #52: how can you speed up lv_arduino ?

simonchen007, 2020-06-13:
hello,

We have a strange issue these days. We try 3 ways for testing LVGL v7 benchmark for esp32.
1) Arduino-esp32 framework + lv_arduino + TFT_eSPI ,  we got about 30 FPS.
2) Esp-idf framework + arduino-esp32(as components) + LVGL（https://github.com/lvgl/lvgl）+ TFT_eSPI, we got about 9 FPS.
3) Lv_port_esp32 + arduino-esp32(as components) + TFT_eSPI, we got about 9 FPS.

As we known, Arduino-esp32 is packaged from  ESP-idf, it should be faster then Arduino-esp32 in  theory. But it's 2 times slower currently.

How can you speed up lv_arduino ?

thanks

Pablo2048, 2020-06-13:
Hi,
I did nothing special so if your measurements are right, then only one thing pops up in my head - loop() task (if the loop() is executed in one task) has higher priority in Arduino-esp32 framework.

simonchen007, 2020-06-14:
so, is that possible to make [Esp-idf framework + arduino-esp32(as components) + LVGL（https://github.com/lvgl/lvgl）+ TFT_eSPI ]  have the same speed ?

Pablo2048, 2020-06-14:
I really don't know - you have to search in arduino-esp32 sources for setup/loop thread parameters and compare it with arduino-esp32 as component. That is all I can suggest...

simonchen007, 2020-06-14:
okay, I will try it. I saw you have esp-id repository too, I guess you have tried that, that's why I ask you .
but of course, I will try to do as you said, 

thanks

closed 2020-07-05 by Pablo2048