lvgl/lv_arduino #56: Embed tft.rect

arcbtc, 2020-07-05:
I was unable to pull in your QRcode extension, but have my own TFT_eSPI tft.rect based function I have used in the past.

How can I embed something like TFT_eSPI tft.rect in an lvgl object like a box?

Here's the function I use for my QRS, it uses another qrcode library 

void showAddress(String XXX){
	  tft.fillScreen(TFT_WHITE);
	  XXX.toUpperCase();
	 const char* addr = XXX.c_str();
	 Serial.println(addr);
	  int qrSize = 12;
	  int sizes[17] = { 14, 26, 42, 62, 84, 106, 122, 152, 180, 213, 251, 287, 331, 362, 412, 480, 504 };
	  int len = String(addr).length();
	  for(int i=0; i<17; i++){
	    if(sizes[i] > len){
	      qrSize = i+1;
	      break;
	    }
	  }
	  QRCode qrcode;
	  uint8_t qrcodeData[qrcode_getBufferSize(qrSize)];
	  qrcode_initText(&qrcode, qrcodeData, qrSize-1, ECC_LOW, addr);
	  Serial.println(qrSize -1);
	 
	  float scale = 2;

	  for (uint8_t y = 0; y < qrcode.size; y++) {
	    for (uint8_t x = 0; x < qrcode.size; x++) {
	      if(qrcode_getModule(&qrcode, x, y)){       
		tft.drawRect(15+2+scale*x, 2+scale*y, scale, scale, TFT_BLACK);
	      }
	      else{
		tft.drawRect(15+2+scale*x, 2+scale*y, scale, scale, TFT_WHITE);
	      }
	    }
	  }
	}

Pablo2048, 2020-07-05:
You can not use low level tft functions directly. You have to put the QR code into LVGL canvas first. I think that somewhere in the forum there is an example how to do it.

closed 2020-07-05 by Pablo2048

arcbtc, 2020-07-06:
Ooooo, thanks!

In case anybody else has a similar issue using the above qrcode library this function works (you'll prob need to change the ```CANVAS_WIDTH``` and ```CANVAS_WIDTH```, and  ``` lv_canvas_set_px(canvas, 15+x*scale+g, 15+y*scale+i, c1);``` the ```15``` adds a 15 pixel boarder to the QR):

            void showQR(lv_obj_t *obj, const char* strtoqr, int qrsize, int scale){
		    #define CANVAS_WIDTH  230
		    #define CANVAS_HEIGHT  230
		    
		    QRCode qrcode;
		    uint8_t qrcodeData[qrcode_getBufferSize(qrsize)];
		    qrcode_initText(&qrcode, qrcodeData, qrsize, ECC_LOW, strtoqr);

		    static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_INDEXED_1BIT(CANVAS_WIDTH, CANVAS_HEIGHT)];
		    lv_obj_t * canvas = lv_canvas_create(obj, NULL);
		    lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_INDEXED_1BIT);
		    lv_canvas_set_palette(canvas, 0, LV_COLOR_BLACK);
		    lv_canvas_set_palette(canvas, 1, LV_COLOR_WHITE);
		    lv_color_t c0;
		    lv_color_t c1;
		    c0.full = 0;
		    c1.full = 1;
		    lv_canvas_fill_bg(canvas, c1, LV_OPA_TRANSP);

		    for (uint8_t y = 0; y < qrcode.size; y++) {
		      for (uint8_t x = 0; x < qrcode.size; x++) {
			if(qrcode_getModule(&qrcode, x, y)){  
			for(int i = 0; i < scale; i++){
			  for(int g = 0; g < scale; g++){
			    lv_canvas_set_px(canvas, 15+x*scale+g, 15+y*scale+i, c0);
			  }
			}
			}
			else{
			for(int i = 0; i < scale; i++){
			  for(int g = 0; g < scale; g++){
			    lv_canvas_set_px(canvas, 15+x*scale+g, 15+y*scale+i, c1);
			  }
			}
			}
		      }
		    }
		}