lvgl/lv_arduino #60: LV_Arduino-3.0.1 esp32 add xpt2046 with a parallel screen

CrocKlok, 2020-11-12:
I had a question I have the ESP32 and have a parallel screen ssd1963 with tft espi working, but I want to add the xpt2046 to work for the example to test out. because tft esp lib can’t handle for now the spi xpt2046 together with a parallel screen. How can I add the touchscreen and how can I calibrated it ? Also I saw that ssd1963 could be use directly in lvgl with the LV_drivers package how does that work. but my main question for now is the xpt2046 can work together with the tft esp parallel screen handeling. 

try to add this lib for xpt2046 from https://github.com/lvgl/lv_drivers but during compiling I got an error in Arduino. lv_arduino-3.0.1\indev\xpt2046.c:13:30: fatal error: drv_XPT2046.h: No such file or directory

What do I miss here? I am now trying the ESP32_TFT_eSPI_slider test

I change this in the initial input device driver
indev_drv.read_cb = XPT2046_read;

I get it as error so I also check in the XPT2046.c but couldn’t find it. So that is why I am also asking it or maybe possible it doesn’t work with it. still unknown for me how to add the xpt2046 driver in the simple slide example.

In file included from C:\Users\XX\AppData\Local\Temp\arduino_modified_sketch_902530\ESP32_TFT_eSPI_Slider.ino:4:0:

c:\users\XX\downloads\arduino-1.8.12\arduino-1.8.12\libraries\lv_arduino-3.0.1\indev\xpt2046.c:13:30: fatal error: drv_XPT2046.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board ESP32 Dev Module.

Already thak you for your reply.

Pablo2048, 2020-11-12:
This repository is obsolete (as written in the readme) so please use lvgl as described here https://github.com/arduino/Arduino/issues/10592 and post your question about touch panel to the forum here https://forum.lvgl.io/ .

closed 2020-11-12 by Pablo2048

CrocKlok, 2020-11-12:
@Pablo2048  I posted a question add https://forum.lvgl.io/ but no response there https://forum.lvgl.io/t/lv-arduino-3-0-1-esp32-add-xpt2046-with-a-parallel-screen/3566/6 so that is why I asked it here.

kisvegabor, 2020-11-13:
@CrocKlok I've answered on the forum.