lvgl/lv_arduino #62: LVGL task Handling

engr-mujahidabdullah, 2020-11-28:
Hi,
I hope all of you are in good health. I am working on ESP32 and SPI_TFT_ILI9341, I my project i have to cover mutliple tasks i.e sensor fusion , tft display, wifi etc etc.. Hence i am at the bigning of my project and i have a problem here i create test task and tft_task in which i call custom functions the problem i that which i observe i that the only task which run is the first one which i call.
[image]

from above figure the only task un is TFT_display 
[image]
 and tft_task is as above 
in next task i only do some serial printing periodically.
Kindly guide me how to manage multiple tasks with lvgl.

embeddedt, 2020-11-28:
If you comment out the LVGL related code does the problem still occur? If so then this is a platform/FreeRTOS issue, not an LVGL one.

engr-mujahidabdullah, 2020-11-28:
I think this issue is partially related to lvgl.

embeddedt, 2020-11-28:
Please check that it happens only when LVGL is in use first. There's not much happening in your code sample, so unless LVGL is corrupting memory somewhere, I'd be surprised if it was causing the problem.

kisvegabor, 2020-11-28:
If `lv_task_handler` halts it's usually becasue of the missing `lv_disp_flush_ready()` in your `disp_flush()` function.
Do you call it?

engr-mujahidabdullah, 2020-11-28:
Yes, I did.  But the matter is that not both tasks run only the first one
which i call run either next_task or tft_task this only happen when i
attach tft_task.

embeddedt, 2020-11-28:
Change `TFT_Task` to this:

```c
void TFT_Task(void *pvParameters)
{
    while(1) {
        vTaskDelay(5);
    }
}
```

See whether the problem still occurs.

Pablo2048, 2020-11-29:
The problem is that the memory allocated for stack is too big IMHO (100000 WORDS!!! - not bytes, for each task) did you check the result of `xTaskCreate` function?

engr-mujahidabdullah, 2020-11-29:
Yes, I did.
However after long debugging I found that the problem is with lv_init()
when i call this function the code blocks or runs only the first task.

Pablo2048, 2020-11-30:
> Yes, I did.
>
And what was the result? I don't believe that the system with 520kB of RAM will allow allocation of 400000 bytes two times... Also I don't know why you are allocating so insane huge stack - I'm using 10 * 1024 and it seems to be more than enough. Because you did not publish the code with `lv_init()` call, I'm unable to help more - just one suggestion: don't forget to use mutex in the task where the `lv_task_handler()` is called (and use this mutex in foreground/second task where it needs to be used). For example I'm doing it like this:
Setup part:
```cpp
  lvglMutex = xSemaphoreCreateMutex();
  xTaskCreate(gui_task, "lvgl", 10 * 1024, NULL, 5, NULL); // Start GUI task
```
and my lvgl handler task looks like:
```cpp
void gui_task(void *dev)
{

  while (1)
  {
    xSemaphoreTake(lvglMutex, portMAX_DELAY);
    lv_task_handler();
    xSemaphoreGive(lvglMutex);
    vTaskDelay(5);
    lv_tick_inc(5);
  }
  vTaskDelete(NULL);
}
```

engr-mujahidabdullah, 2020-11-30:
Thanks to all,
I resolved the issue by initializing more tasks. Is any example of auto
scrolling of textbox when keyboard popup.

embeddedt, 2020-11-30:
`lv_demo_widgets` does this by [resizing the tabview](https://github.com/lvgl/lv_examples/blob/25ae1dc9842b23d3c27effcd1b1a98c7e8688151/src/lv_demo_widgets/lv_demo_widgets.c#L562-L593) to be half the screen height when the keyboard is created. When the keyboard is deleted, the tabview is returned to its old size.

You can see `lv_demo_widgets` online [here](https://lvgl.github.io/lv_examples/lv_demo_widgets/index.html).

closed 2020-12-01 by Pablo2048